tensorflow/tflite-micro #2246: Where is array.h file ?

SahinTiryaki, 2023-09-26:
[image]

array.h file is not founded ? Can u help me please ?

lexcsam, 2023-09-27:
For some reasons they aren't auto pulled in write some code to copy them in so you need to copy the array.h files into the place in your build and make sure they are in your cmake. there was a similar issue before , take a look there #2078

rascani, 2023-09-27:
The [tensorflow/lite/array.h](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/array.h) file is in the tree, but we don't actually want to use it in TFLite-Micro. Rather than copying the files in the tree, you should add a define for `TF_LITE_STATIC_MEMORY` to your compilation rules. Please note that the [aforementioned kernel_util.cc has an ifdef guard  around the array.h include](https://github.com/tensorflow/tflite-micro/blob/266d49d09ca2cce3def9edd87530c3d3bb88e27d/tensorflow/lite/kernels/kernel_util.cc#L25). Our makefiles include this preprocessor define [here](https://github.com/tensorflow/tflite-micro/blob/266d49d09ca2cce3def9edd87530c3d3bb88e27d/tensorflow/lite/micro/tools/make/Makefile#L161).

closed 2023-09-27 by rascani