tensorflow/tflite-micro #2261: TensorflowLite error: Didn't find op for builtin opcode 'WHERE' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

tjdwo1105, 2023-10-05:
Hello,

I would like to run the model that I made on the Arduino BLE 33 board(nRF52840)
while uploading to board using ArduinoIDE below massage appeared

> Didn't find op for builtin opcode 'WHERE' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?
> Failed to get registration from op code WHERE
> AllocateTensors() failed
> Invoke failed!

**Does anyone know how to solve the problem?**
I am new to TensorFlow and studying it. Please someone who has gone through help me.

------
**my version status**
tensorflow==2.8.0
TensorFlow Lite Micro Library `git clone https://github.com/tensorflow/tflite-micro-arduino-examples Arduino_TensorFlowLite`
Arduino IDE 2.2.1

**What I been through..**
I converted to .tflite file and changed to model.h and model.cc with below code
```
!echo "const unsigned char model[] = {" > /content/model.h
!cat gesture_model.tflite | xxd -i      >> /content/model.h
!echo "};"                              >> /content/model.h

import os
model_h_size = os.path.getsize("model.h")
print(f"Header file, model.h, is {model_h_size:,} bytes.")
print("\nOpen the side panel (refresh if needed). Double click model.h to download the file.")
```
**Trial 0) Using `AllOpsResolver`**
The error appeared.
> Didn't find op for builtin opcode 'WHERE' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?
> Failed to get registration from op code WHERE
> AllocateTensors() failed
> Invoke failed!

**Trial 1) Using `MicroMutableOpResolver` instead of `AllOpsResolver`**
However, there is no `AddWhere()` function to add the WHERE op to the MicroMutableOpResolver. 
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_mutable_op_resolver.h
```
static tflite::MicroMutableOpResolver<17> micro_op_resolver;
  micro_op_resolver.AddPad();
  micro_op_resolver.AddReshape();
  micro_op_resolver.AddConv2D();
  micro_op_resolver.AddTranspose();
  micro_op_resolver.AddMaxPool2D();
  micro_op_resolver.AddWhile();
  micro_op_resolver.AddConcatenation();
  micro_op_resolver.AddEqual();
  //micro_op_resolver.AddWhere();  // no function
  micro_op_resolver.AddGather();
  micro_op_resolver.AddShape();
  micro_op_resolver.AddCast();
  //micro_op_resolver.AddSparseToDense();  // no function
  micro_op_resolver.AddAdd();
  micro_op_resolver.AddStridedSlice();
  micro_op_resolver.AddPack();
  micro_op_resolver.AddFullyConnected();
  micro_op_resolver.AddLess();
  micro_op_resolver.AddIf();

// Build an interpreter to run the model with.
  static tflite::MicroInterpreter static_interpreter(
    model, micro_op_resolver, tensor_arena, kTensorArenaSize);
```
The following error masseges printed
>MicroAllocator: Model allocation started before finishing the previously allocated model
>Failed starting model allocation.
>Invoke failed!

FYI) The model uses 19 OPs as shown in the above-mentioned code.

**Trial 2) Does TF Lite not support the WHERE op?**
It seems that the answer is NO. If you see the below link, there is the explanation of tfl.where and tfl.sparse_to_dense.
https://www.tensorflow.org/mlir/tfl_ops?hl=ko#tflwhere_tflwhereop

**Trial 3) The OP Version Problem?**
I've checked the op version info of this link
https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/tools/versioning/runtime_version.cc
```
{{BuiltinOperator_WHERE, 1}, "1.14.0"}
{{BuiltinOperator_WHERE, 2}, "2.8.0"},
```
I think this is the problem. However, I can not find a way to modify the version that the model uses.

It would be helpful if someone comments about what I did. Thankyou!

Thank you in advance.

tjdwo1105, 2023-10-30:
> "This issue is being marked as stale due to inactivity. Remove label or comment to prevent closure in 5 days."

I am still working on it.

rascani, 2023-10-30:
TFLite for Microcontrollers only supports a subset of TFLite ops. It does not currently support the WHERE or SPARSE_TO_DENSE op. Your options are to change the model to avoid those operations, or you can port the ops to TFLM. There is a guide on how to do so here: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md

0xcabrex, 2023-11-26:
I have a similar error that goes like this:

Didn't find op for builtin opcode 'SHAPE' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

I am running my tflite model on an esp32-s3 board. Do not have a solution right now. 

the [list](https://www.tensorflow.org/lite/guide/op_select_allowlist) of supported op codes mention the shape attribute, yet I am getting the error. What could be the solution?

closed 2023-12-28 by github-actions[bot]