tensorflow/tflite-micro #2262: Missing support for Pow op

blaine-rister, 2023-10-05:
It looks like there is currently no kernel for the `tfl.pow` op. This op is useful for some audio tasks like speech denoising, where it can compress the dynamic range of the input spectrogram. See this paper as an example: https://arxiv.org/abs/2005.11138

Would it be difficult to add `tfl.pow` support? 

This would be very helpful to the MLPerf Tiny working group which is trying to use TFLM as the reference solution for a new speech denoising benchmark.

blaine-rister, 2023-11-03:
Bumping to keep this alive

rascani, 2023-11-03:
The guide on how to port TFLite ops to TFLM is located here:
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md

It is not particularly difficult, but can be time consuming.