tensorflow/tflite-micro #2288: create_tflm_tree.py fails with "make: tensorflow/lite/micro/tools/make/flatbuffers_download.sh: Command not found"

edwillys, 2023-10-25:
Hello,

I'm trying to start the port of tflite-micro to a new architecture. By following the instructions, I tried to run:

`python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py -e hello_world ./tflm-tree`

To which it fails:

```
Traceback (most recent call last):
  File "tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py", line 303, in <module>
    main()
  File "tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py", line 280, in main
    (" ".join(params_list), stderr.decode()))
RuntimeError: make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads failed with

 make: tensorflow/lite/micro/tools/make/flatbuffers_download.sh: Command not found
tensorflow/lite/micro/tools/make/Makefile:586: *** Something went wrong with the flatbuffers download: .  Stop.
```

I checked and the flatbuffers_download.sh is not present under tensorflow/lite/micro/tools/make/. Am I missing something?

Thanks in advance!

rascani, 2023-10-25:
You may need to re-sync your repository. The flatbuffers_download.sh script is indeed located at tensorflow/lite/micro/tools/make/

https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/make/flatbuffers_download.sh

closed 2023-10-25 by rascani

edwillys, 2023-10-26:
Thanks. Indeed the solution was to delete the folder and re-clone it within WSL (as I was running on Windows)