tensorflow/tflite-micro #2289: could it be used in cortex-A processor?

zrx563010758, 2023-10-26:
I'm trying to use TFLM in Cortex-A processor, but I have no idea how to make it. It seems like I need to use the TensorFlow folder(include core and lite)and the third party(I found in [https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md]), have any suggestions?

rascani, 2023-10-26:
We don't have any optimized kernels for Cortex-A, so you'll have to just stick with the reference kernels. If you follow the instructions for [building the static library with reference kernels](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md#step-1-build-tflm-static-library-with-reference-kernels), you should be able to create a source tree with just the sources you need.

There's also a basic makefile located [here](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/project_generation/Makefile) that should be enough to get you started. You'll need to modify the compiler variables (CXX, AR, etc). to use your toolchain.

zrx563010758, 2023-10-27:
I followed the instruction(https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md#step-1-build-tflm-static-library-with-reference-kernels) on my MacBook Pro M1 Pro chip, used the code in the terminal :
```
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py \
  -e hello_world \
  -e micro_speech \
  -e person_detection \
  /tmp/tflm-tree
```
it returns:
```
Traceback (most recent call last):
  File "/Users/zrx/Documents/tflite-micro/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py", line 303, in <module>
    main()
  File "/Users/zrx/Documents/tflite-micro/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py", line 279, in main
    raise RuntimeError("%s failed with \n\n %s" %
RuntimeError: make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads failed with 

 tensorflow/lite/micro/tools/make/flatbuffers_download.sh: line 63: wget: command not found
tensorflow/lite/micro/tools/make/Makefile:589: *** Something went wrong with the flatbuffers download: .  Stop.
```
I don't know how to fix it,do you know where goes wrong ?

zrx563010758, 2023-10-27:
And I tried using the code:
`make -f tensorflow/lite/micro/tools/make/Makefile generate_projects`
it returns:
```
(base) zrx@zrxdeMacBook-Pro tflite-micro % make -f tensorflow/lite/micro/tools/make/Makefile generate_projects
tensorflow/lite/micro/tools/make/flatbuffers_download.sh: line 63: wget: command not found
tensorflow/lite/micro/tools/make/Makefile:589: *** Something went wrong with the flatbuffers download: .  Stop.
```
and I sure I have the flatbuffers_download.sh file . Can you help me?

jonnor, 2023-11-14:
You are missing the program `wget`, as the error says.
```
tensorflow/lite/micro/tools/make/flatbuffers_download.sh: line 63: wget: command not found
```

closed 2023-12-15 by github-actions[bot]