tensorflow/tflite-micro #2290: Input Tensor Reuse after inference

XenuIsWatching, 2023-10-26:
After the data is copied in to the input tensor and then inference is invoked. The original input tensor no longer contains the original data. In other words, if an image data is copied in to the input tensor, and then the data is read back from where the input tensor was, the data does not match and not be further reused as the original input data has now been corrupted after inference was ran.

Why is the input tensor being overwritten the model. If possible, is there a way to have the inference leave the input tensor as is?

rascani, 2023-10-26:
By default, the `MicroAllocator` is constructed with the `GreedyMemoryPlanner`, which re-uses memory once tensors are no longer using it. In other words, once that input tensor is done being used, the backing memory for it gets re-purposed for another tensor. This is done to minimize the amount of memory needed for the arena. 

It is possible to construct a `MicroAllocator` with the `LinearMemoryPlanner`, which will lay out all tensors separately. That, of course, also means the amount of memory necessary for the arena will increase.

XenuIsWatching, 2023-10-29:
> By default, the `MicroAllocator` is constructed with the `GreedyMemoryPlanner`, which re-uses memory once tensors are no longer using it. In other words, once that input tensor is done being used, [...]

I see. I was not aware of that. Do you know where I can find documentation on these memory planners (including HillClimb?)

rascani, 2023-10-30:
You can find a few memory management related docs here: https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/docs

Most of the documentation on the memory planners themselves are located with the code here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/memory_planner

XenuIsWatching, 2023-11-26:
It appears that the `LinearMemoryPlanner` will greatly increase the memory neccessary. Would there be a way to just have the input tensor not be 'dealloced' (even if that means doing something hacky in the code?

rascani, 2023-11-27:
You might be able to do this by modifying `AllocationInfoBuilder::MarkAllocationLifetimes` function to ensure that the `AllocationInfo::last_used` gets maxed out for input tensors. We do this for output tensors to ensure they're persistent:

https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_allocation_info.cc#L321

I'd probably try to copy that loop but for subgraph inputs and remove the `UpdateFirstCreated()` call. Note that this function is called for each subgraph, so you might want to conditionalize it for just the first subgraph.

closed 2023-12-28 by github-actions[bot]