tensorflow/tflite-micro #2291: third_party/gemmlowp/fixedpoint/fixedpoint.h:520:27: error: no matching function for call to 'max(long int, const double&)' min_bound),

eppane, 2023-10-27:
Hello,

I am trying to run the hello_world example on Cortex-M4F-equipped [Adafruit Metro M4 Express](https://circuitpython.org/board/metro_m4_express/), but I am getting the following error:

```
In file included from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/common.h:28,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/reference/integer_ops/pooling.h:21,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/kernels/pooling.h:23,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:32,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/all_ops_resolver.h:19,
                 from C:\Users\abc\AppData\Local\Temp\.arduinoIDE-unsaved2023926-7172-jv6p6q.i19h\hello_world\hello_world.ino:22:
c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/gemmlowp/fixedpoint/fixedpoint.h: In static member function 'static gemmlowp::FixedPoint<tRawType, tIntegerBits> gemmlowp::FixedPoint<tRawType, tIntegerBits>::FromDouble(double)':
c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/gemmlowp/fixedpoint/fixedpoint.h:520:27: error: no matching function for call to 'max(long int, const double&)'
  520 |                  min_bound),
      |                           ^
In file included from c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\bits\char_traits.h:39,
                 from c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\string:40,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/base.h:51,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/array.h:20,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/flatbuffers.h:22,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/schema/schema_generated.h:7,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/core/api/flatbuffer_conversions.h:28,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:22,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/all_ops_resolver.h:19,
                 from C:\Users\abc\AppData\Local\Temp\.arduinoIDE-unsaved2023926-7172-jv6p6q.i19h\hello_world\hello_world.ino:22:
c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\bits\stl_algobase.h:222:5: note: candidate: 'template<class _Tp> const _Tp& std::max(const _Tp&, const _Tp&)'
  222 |     max(const _Tp& __a, const _Tp& __b)
      |     ^~~
c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\bits\stl_algobase.h:222:5: note:   template argument deduction/substitution failed:
In file included from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/common.h:28,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/reference/integer_ops/pooling.h:21,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/kernels/pooling.h:23,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:32,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/all_ops_resolver.h:19,
                 from C:\Users\abc\AppData\Local\Temp\.arduinoIDE-unsaved2023926-7172-jv6p6q.i19h\hello_world\hello_world.ino:22:
c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/gemmlowp/fixedpoint/fixedpoint.h:520:27: note:   deduced conflicting types for parameter 'const _Tp' ('long int' and 'double')
  520 |                  min_bound),
      |                           ^
In file included from c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\bits\char_traits.h:39,
                 from c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\string:40,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/base.h:51,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/array.h:20,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/flatbuffers.h:22,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/schema/schema_generated.h:7,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/core/api/flatbuffer_conversions.h:28,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:22,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/all_ops_resolver.h:19,
                 from C:\Users\abc\AppData\Local\Temp\.arduinoIDE-unsaved2023926-7172-jv6p6q.i19h\hello_world\hello_world.ino:22:
c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\bits\stl_algobase.h:268:5: note: candidate: 'template<class _Tp, class _Compare> const _Tp& std::max(const _Tp&, const _Tp&, _Compare)'
  268 |     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
      |     ^~~
c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\bits\stl_algobase.h:268:5: note:   template argument deduction/substitution failed:
In file included from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/common.h:28,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/reference/integer_ops/pooling.h:21,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/kernels/pooling.h:23,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:32,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/micro/all_ops_resolver.h:19,
                 from C:\Users\abc\AppData\Local\Temp\.arduinoIDE-unsaved2023926-7172-jv6p6q.i19h\hello_world\hello_world.ino:22:
c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/gemmlowp/fixedpoint/fixedpoint.h:520:27: note:   deduced conflicting types for parameter 'const _Tp' ('long int' and 'double')
  520 |                  min_bound),
      |                           ^
In file included from c:\users\abc\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\c++\9.2.1\algorithm:62,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/base.h:55,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/array.h:20,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/third_party/flatbuffers/include/flatbuffers/flatbuffers.h:22,
                 from c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/schema/schema_generated.h:7,
[... cut ...]
c:\Users\abc\Documents\Arduino\libraries\Arduino_TensorFlowLite\src/tensorflow/lite/kernels/internal/cppmath.h:35:1: note: in expansion of macro 'DECLARE_STD_GLOBAL_SWITCH1'
   35 | DECLARE_STD_GLOBAL_SWITCH1(TfLiteRound, round);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

Any idea what could be the cause? Seems to be related to how pooling is implemented in `gemmlowp`.

Otherwise I just pulled the latest `Arduino_TensorFlowLite` and I am using Arduino IDE 2.2.1. I used to be able to run TFLM on Metro M4 Express still about a year and a half ago.

Thanks!

ddavis-2015, 2023-10-27:
@eppane 

Could you give additional detail on where you downloaded the TFLM codebase from?  Did you clone the [Arduino examples repo](https://github.com/tensorflow/tflite-micro-arduino-examples)?  That repo is up to date as of March 2023 and is the only official source of TFLM for Arduino code.  It supports the Arduino Nano 33 BLE (NRF52840, cortex M4).

eppane, 2023-10-29:
Hello @ddavis-2015!

Yes, I cloned that one. It says that it should be compatible with most Arm Cortex M-based boards and regarding peripherals, I commented out L61 `#error "unsupported board"` in `src\peripherals\peripherals.h`. The hello world example is not using any Nano-specific peripherals, right?

Here is a code snippet I have successfully ran before and I am now trying to modify it to support the current codebase:

```
#include <TensorFlowLite.h>
#include "tensorflow/lite/micro/all_ops_resolver.h"
#include "tensorflow/lite/micro/micro_error_reporter.h"
#include "tensorflow/lite/micro/micro_interpreter.h"
#include "tensorflow/lite/schema/schema_generated.h"
#include "tensorflow/lite/version.h"

// Model
#include "6_v16.h"

// Globals, used for compatibility with Arduino-style sketches.
namespace {
tflite::ErrorReporter* error_reporter = nullptr;
const tflite::Model* model = nullptr;
tflite::MicroInterpreter* interpreter = nullptr;
TfLiteTensor* input = nullptr;
TfLiteTensor* output = nullptr;

constexpr int kTensorArenaSize = 50*1024;
uint8_t tensor_arena[kTensorArenaSize];

constexpr int N_INPUTS = 65*4;
constexpr int N_OUTPUTS = 1;
}

float X_test[N_INPUTS] = {0.09090899676084518,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.09090899676084518,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.15556000173091888,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.15556000173091888,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.15556000173091888,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.15556000173091888,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.15556000173091888,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.11111000180244446,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.13636000454425812,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.022221999242901802,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.2272700071334839,0.022221999242901802,-0.20000000298023224,-0.2340400069952011,0.2272700071334839,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.2272700071334839,0.022221999242901802,-0.20000000298023224,-0.2340400069952011,0.2272700071334839,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.2272700071334839,0.022221999242901802,-0.20000000298023224,-0.2340400069952011,0.2272700071334839,0.022221999242901802,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.2444400042295456,-0.2340400069952011,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2766000032424927,0.18182000517845154,0.06666699796915054,-0.20000000298023224,-0.2766000032424927,0.13636000454425812,0.06666699796915054,-0.20000000298023224,-0.2766000032424927,0.13636000454425812,-0.022221999242901802,-0.20000000298023224,-0.4042600095272064,0.09090899676084518,-0.15556000173091888,-0.2444400042295456,-0.4042600095272064,0.09090899676084518,-0.06666699796915054,-0.28889000415802,-0.5744699835777283,0.09090899676084518,-0.06666699796915054,-0.3333300054073334,-0.4468100070953369,0.09090899676084518,0.022221999242901802,-0.3333300054073334,-0.4468100070953369,-0.045455001294612885,0.06666699796915054,-0.3333300054073334,-0.4468100070953369,-0.045455001294612885,0.06666699796915054,-0.3333300054073334,-0.4893600046634674,0.0,0.11111000180244446,-0.3777799904346466,-0.4893600046634674,0.0,0.20000000298023224,-0.3777799904346466,-0.5319100022315979,-0.31817999482154846,0.20000000298023224,-0.3333300054073334,-0.5319100022315979,0.09090899676084518,0.28889000415802,-0.28889000415802,-0.5319100022315979,0.13636000454425812,0.28889000415802,-0.28889000415802,-0.5319100022315979,0.13636000454425812,0.28889000415802,-0.3333300054073334,-0.10638 [... cut ...]
                         static_cast<float>(X_test[0]));
    return;
  }

  // Dequantize output
  for (int i = 0; i < N_OUTPUTS; i++) {
    y[i] = (float)((output->data.int8[i] - output->params.zero_point) * output->params.scale);
  }

  inference_count += 1;
}

void showResults() {
  Serial.print("PROBABILITY: ");
  Serial.println(y[0]);
  Serial.print("\n\n");
}

```

How would you go from here, any ideas? I took away the micro error reporter and added MicroPrintf, modifying accordingly. And also the `version.h`.

Thank you!

ddavis-2015, 2023-11-01:
@eppane 

The compilation issues you face with the hello_world example are two-fold:
* The repo is currently only supporting the Arduino Nano 33 BLE, and does not compile without modification on any other platform
* Arduino Board Support Packages (BSPs) often contain macros for common functions which act in direct conflict with their C++11 standard library equivalents.  You can see this in the peripherals.h header file with the need to `#undef abs`.  In your case, the Adafruit BSP defines a `round` macro, making the TFLM code unable to compile correctly.

The best way to work around these issues is as follows:
1) remove all files from src/peripherals except for the `peripherals.h` and `utility.h` files.
1) modify src/peripherals/peripherals.h to change the lines:
```
#else  // ARDUINO_ARDUINO_NANO33BLE
#error "unsupported board"
```
and replace them with:
```
#elif defined(ADAFRUIT_METRO_M4_EXPRESS)

// Temporary fix, similar to buganizer #268498682, arduino-examples issue #169
#undef round

#else  // ARDUINO_ARDUINO_NANO33BLE
#error "unsupported board"
```
1) modify the file src/tensorflow/lite/micro/system_setup.cpp, replacing this line:
```
#if defined(ARDUINO) && !defined(ARDUINO_ARDUINO_NANO33BLE)
```
with the following line:
```
#if defined(ARDUINO) && !defined(ADAFRUIT_METRO_M4_EXPRESS)
```

Making these changes should allow the hello_world example to compile and link correctly.  Tested with Arduino IDE 1.8.19 and Adafruit SAMD BSP 1.7.13 versions.

As this is an unsupported platform, no guarantee can be made that hello_world runs correctly on your hardware.

This issue will now be closed.  Please feel free to reopen the issue should you wish to provide further comments.

closed 2023-11-01 by ddavis-2015

eppane, 2023-11-02:
Hello @ddavis-2015!

Thank you for your expertise and for these suggestions, worked like a charm! 🙏