tensorflow/tflite-micro #2293: ESP32 still supported?

h3ndrik, 2023-10-30:
Since `TARGET=esp` doesn't seem to be valid any more, I've tried building the example with:

`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=xtensa ...`

That returns a `tensorflow/lite/micro/tools/make/targets/xtensa_makefile.inc:18: *** XTENSA_BASE is undefined.  Stop.`
And indeed, in _tensorflow/lite/micro/tools/make/targets/xtensa_makefile.inc_ it says:

```
#   - XTENSA_BASE  must be set to location of
#     the Xtensa developer tools installation directory.
```

But what are the "Xtensa developer tools"? Are they contained somewhere within the ESP-IDF, or do I need to apply/buy a proprietary SDK somewhere? I've tried to read the documentation of tflite-micro but didn't find any up-to-date build instructions for this platform.

(I'm trying to build some Arduino libraries or the speech example project for an ESP32 board.)

rascani, 2023-10-30:
We don't maintain the ESP32 platform. I would generally assume that there is an xtensa toolchain as part of ESP-IDF.

You may want to use create_tflm_tree.py and setup a clean tree, and use the basic makefile located there.
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/tools/project_generation

h3ndrik, 2023-10-31:
Thanks. TFLite Micro really changed since I've last tinkered around with it.

Is there a way to incorporate the optimized kernel implementations?

```bash
$ python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py -e micro_speech --makefile_options="TARGET=xtensa OPTIMIZED_KERNEL_DIR=xtensa TARGET_ARCH=project_generation" ./tflm-out
[...]
tensorflow/lite/micro/tools/make/targets/xtensa_makefile.inc:18: *** XTENSA_BASE is undefined.  Stop.
```

But at least without them, it generates the code.

rascani, 2023-10-31:
The `OPTIMIZED_KERNEL_DIR` should be sufficient enough to pick up the optimized kernel implementations. You can probably just give a dummy path for `XTENSA_BASE` to make the script happy. You'll probably also need `XTENSA_TOOLS_VERSION` and `XTENSA_CORE` as well.

h3ndrik, 2023-10-31:
Thanks for the help! I'm going to be optimistic and close this issue.

```bash
XTENSA_BASE="none" XTENSA_TOOLS_VERSION="none" XTENSA_CORE="none" python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py -e micro_speech --makefile_options="TARGET=xtensa OPTIMIZED_KERNEL_DIR=xtensa TARGET_ARCH=hifimini" ./tflm-out
```

seems to work. I'm not sure what the TARGET_ARCH does, but guess I'll find out. I just found how [espressif does it](https://github.com/espressif/esp-tflite-micro/blob/master/scripts/sync_from_tflite_micro.sh). Maybe I can have a look at their customizations.

closed 2023-10-31 by h3ndrik