tensorflow/tflite-micro #2319: Dilated Convolutional Layers Produce Wrong Output

apage224, 2023-11-16:
Issue: Models containing dilated convolutional layers produce incorrect output. This is true for 1D and 2D convolutional layers, all tested dilation_rate sizes, and for both fp32 and int8 quantized models. This issue has been brought up in #1646, #1629, #1231 which had no response nor resolution and closed to due being stale.

Reproduce: Running the following script will show that TF and TFLite produce correct results while TFLM does not.

```python

import numpy as np
import tensorflow as tf
import tflite_micro as tflm

INPUT_SHAPE = (8, 1)
DILATION_RATE = 2  # <- If 1 then assertions pass

# Create example dilated convolutional model
inputs = tf.keras.layers.Input(shape=INPUT_SHAPE)
y = inputs
y = tf.keras.layers.Convolution1D(1, 3, dilation_rate=DILATION_RATE, padding='same')(y)
model = tf.keras.Model(inputs=inputs, outputs=y)

# Convert to TF Lite model
converter = tf.lite.TFLiteConverter.from_keras_model(model=model)
tflite_model = converter.convert()

# Create TFLite and TFLM interpreter
tfl_interpreter = tf.lite.Interpreter(model_content=tflite_model)
tfl_interpreter.allocate_tensors()

tflm_interpreter = tflm.runtime.Interpreter.from_bytes(tflite_model)

# Create input tensor of 1s
x = np.ones((1,)+INPUT_SHAPE, dtype=np.float32)

# Invoke TF, TFLite and TFLM models using same input
y_tf = model(x)

tfl_interpreter.set_tensor(tfl_interpreter.get_input_details()[0]["index"], x)
tfl_interpreter.invoke()
y_tfl = tfl_interpreter.get_tensor(tfl_interpreter.get_output_details()[0]["index"])

tflm_interpreter.set_input(x, 0)
tflm_interpreter.invoke()
y_tflm = tflm_interpreter.get_output(0)

# Below passes
np.testing.assert_almost_equal(y_tf, y_tfl, decimal=3, err_msg="TF and TFLite output do not match.")

# Below fails
np.testing.assert_almost_equal(y_tf, y_tflm, decimal=3, err_msg="TF and TFLM output do not match.")

```

ddavis-2015, 2023-11-17:
@apage224 

Thank you for your interest in TFLM.  Your reporting of this issue is much appreciated.  This issue is currently under investigation.

apage224, 2023-11-17:
Thanks for the prompt response!

Just some additional findings. I used the TFLite analyzer to compare intermediate tensors using below code. From what I can see the problem seems to start at operation `SpaceToBatchND`. I believe the TFL and TFLM use the exact same op implementation for this. It appears, however, the TFLM constructs incorrect tensor shape. I've included output from trivial example above. 

``` python

tf.lite.experimental.Analyzer.analyze(model_content=tflite_model)

for i in range(0, tfl_interpreter._interpreter.NumTensors(0)):
    print(f"Tensor {i}: {tfl_interpreter.get_tensor_details()[i]['name']}")
    tflm_tensor = tflm_interpreter.GetTensor(i, 0)["tensor_data"]
    try:
        tfl_tensor = tfl_interpreter.get_tensor(i, 0)
    except ValueError:
        print("  TFL: N/A")
        print(f" TFLM: shape={tflm_tensor.shape}, dtype={tflm_tensor.dtype}")
        print("")
        continue

    is_match = np.allclose(tfl_tensor, tflm_tensor, atol=1e-3)
    print(f"  TFL: shape={tfl_tensor.shape}, dtype={tfl_tensor.dtype}")
    print(f" TFLM: shape={tflm_tensor.shape}, dtype={tflm_tensor.dtype}")
    print(f" MATCH: {'YES' if is_match else 'NO'}")
    print("")

```

__Console Output__:
``` console

Number of resource variables the model uses =  0
=== TFLite ModelAnalyzer ===

Your TFLite model has '1' subgraph(s). In the subgraph description below,
T# represents the Tensor numbers. For example, in Subgraph#0, the SPACE_TO_BATCH_ND op takes
tensor #0 and tensor #4 and tensor #7 as input and produces tensor #8 as output.

Subgraph#0 main(T#0) -> [T#13]
  Op#0 SPACE_TO_BATCH_ND(T#0, T#4[2], T#7[2, 2]) -> [T#8]
  Op#1 EXPAND_DIMS(T#8, T#6[-3]) -> [T#9]
  Op#2 CONV_2D(T#9, T#3, T#2) -> [T#10]
  Op#3 RESHAPE(T#10, T#1[-1, 4, 1]) -> [T#11]
  Op#4 BATCH_TO_SPACE_ND(T#11, T#4[2], T#5[0, 0]) -> [T#12]
  Op#5 ADD(T#12, T#2) -> [T#13]

Tensors of Subgraph#0
  T#0(serving_default_input_1:0) shape_signature:[-1, 8, 1], type:FLOAT32
  T#1(model/conv1d/Conv1D/Squeeze) shape:[3], type:INT32 RO 12 bytes, buffer: 2, data:[-1, 4, 1]
  T#2(model/conv1d/BiasAdd/ReadVariableOp) shape:[1], type:FLOAT32 RO 4 bytes, buffer: 3, data:[0]
  T#3(model/conv1d/Conv1D) shape:[1, 1, 3, 1], type:FLOAT32 RO 12 bytes, buffer: 4, data:[0.400634, 0.725455, 0.995139]
  T#4(model/conv1d/Conv1D/BatchToSpaceND/block_shape) shape:[1], type:INT32 RO 4 bytes, buffer: 5, data:[2]
  T#5(model/conv1d/Conv1D/BatchToSpaceND/crops) shape:[1, 2], type:INT32 RO 8 bytes, buffer: 6, data:[0, 0]
  T#6(model/conv1d/Conv1D/ExpandDims/dim) shape:[], type:INT32 RO 4 bytes, buffer: 7, data:[-3]
  T#7(model/conv1d/Conv1D/SpaceToBatchND/paddings) shape:[1, 2], type:INT32 RO 8 bytes, buffer: 8, data:[2, 2]
  T#8(model/conv1d/Conv1D/SpaceToBatchND) shape_signature:[-1, 6, 1], type:FLOAT32
  T#9(model/conv1d/Conv1D/ExpandDims) shape_signature:[-1, 1, 6, 1], type:FLOAT32
  T#10(model/conv1d/Conv1D1) shape_signature:[-1, 1, 4, 1], type:FLOAT32
  T#11(model/conv1d/Conv1D/Squeeze1) shape_signature:[-1, 4, 1], type:FLOAT32
  T#12(model/conv1d/Conv1D/BatchToSpaceND) shape_signature:[-1, 8, 1], type:FLOAT32
  T#13(StatefulPartitionedCall:0) shape_signature:[-1, 8, 1], type:FLOAT32

---------------------------------------------------------------
Your TFLite model has '1' signature_def(s).

Signature#0 key: 'serving_default'
- Subgraph: Subgraph#0
- Inputs: 
    'input_1' : T#0
- Outputs: 
    'conv1d' : T#13

---------------------------------------------------------------
              Model size:       2616 bytes
    Non-data buffer size:       2464 bytes (94.19 %)
  Total data buffer size:        152 bytes (05.81 %)
    (Zero value buffers):         12 bytes (00.46 %)

* Buffers of TFLite model are mostly used for constant tensors.
  And zero value buffers are buffers filled with zeros.
  Non-data buffers area are used to store operators, subgraphs and etc.
  You can find more details from https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/schema/schema.fbs

Tensor 0: serving_default_input_1:0
  TFL: shape=(1, 8, 1), dtype=float32
 TFLM: shape=(1, 8, 1), dtype=float32
 MATCH: YES

Tensor 1: model/conv1d/Conv1D/Squeeze
  TFL: shape=(3,), dtype=int32
 TFLM: shape=(3,), dtype=int32
 MATCH: YES

Tensor 2: model/conv1d/BiasAdd/ReadVariableOp
  TFL: shape=(1,), dtype=float32
 TFLM: shape=(1,), dtype=float32
 MATCH: YES

Tensor 3: model/conv1d/Conv1D
  TFL: shape=(1, 1, 3, 1), dtype=float32
 TFLM: shape=(1, 1, 3, 1), dtype=float32
 MATCH: YES

Tensor 4: model/conv1d/Conv1D/BatchToSpaceND/block_shape
  TFL: shape=(1,), dtype=int32
 TFLM: shape=(1,), dtype=int32
 MATCH: YES

Tensor 5: model/conv1d/Conv1D/BatchToSpaceND/crops
  TFL: shape=(1, 2), dtype=int32
 TFLM: shape=(1, 2), dtype=int32
 MATCH: YES

Tensor 6: model/conv1d/Conv1D/ExpandDims/dim
  TFL: shape=(), dtype=int32
 TFLM: shape=(), dtype=int32
 MATCH: YES

Tensor 7: model/conv1d/Conv1D/SpaceToBatchND/paddings
  TFL: shape=(1, 2), dtype=int32
 TFLM: shape=(1, 2), dtype=int32
 MATCH: YES

Tensor 8: model/conv1d/Conv1D/SpaceToBatchND
  TFL: shape=(2, 6, 1), dtype=float32
 TFLM: shape=(1, 6, 1), dtype=float32
 MATCH: NO

Tensor 9: model/conv1d/Conv1D/ExpandDims
  TFL: shape=(2, 1, 6, 1), dtype=float32
 TFLM: shape=(1, 1, 6, 1), dtype=float32
 MATCH: NO

Tensor 10: model/conv1d/Conv1D1
  TFL: N/A
 TFLM: shape=(1, 1, 4, 1), dtype=float32

Tensor 11: model/conv1d/Conv1D/Squeeze1
  TFL: shape=(2, 4, 1), dtype=float32
 TFLM: shape=(1, 4, 1), dtype=float32
 MATCH: NO

Tensor 12: model/conv1d/Conv1D/BatchToSpaceND
  TFL: shape=(1, 8, 1), dtype=float32
 TFLM: shape=(1, 8, 1), dtype=float32
 MATCH: NO

Tensor 13: StatefulPartitionedCall:0
  TFL: shape=(1, 8, 1), dtype=float32
 TFLM: shape=(1, 8, 1), dtype=float32
 MATCH: NO

```

ddavis-2015, 2023-11-19:
@apage224 

The level of detail you have provided is really appreciated.

Further investigation shows that Keras models when used with the `TfLiteConverter`, may generate unexpected (but correct) graphs.  In your case the dilation parameter is removed and additional layers are generated to compensate for the dilation (as is shown by the `Analyzer`).  It is possible for the converter to generate a model closer to your intent by using the low level TensorFlow APIs:

```
# Create example dilated convolutional model
class ConvModel(tf.Module):
  def __init__(self, **kwargs):
    super().__init__(**kwargs)
    # random, just like Keras uses for initial filter values
    self._filters = tf.constant([0.812782, -0.248076, -0.0599043], shape=(3, 1, 1))

  def __call__(self, x):
    return tf.nn.conv1d(x,
                        filters=self._filters, 
                        stride=1, 
                        padding="SAME",
                        dilations=DILATION_RATE)

model = ConvModel()
model_func = tf.function(func=model)
cf = model_func.get_concrete_function(
          tf.TensorSpec(shape=(1,)+INPUT_SHAPE, dtype=tf.float32))

# Convert to TF Lite model
converter = tf.lite.TFLiteConverter.from_concrete_functions([cf], model)
tflite_model = converter.convert()
```
This will generate the following from the `Analyzer`:
```
Subgraph#0 main(T#0) -> [T#7]
  Op#0 RESHAPE(T#0, T#1[1, 1, 8, 1]) -> [T#5]
  Op#1 CONV_2D(T#5, T#4, T#3) -> [T#6]
  Op#2 RESHAPE(T#6, T#2[1, 8, 1]) -> [T#7]

Tensors of Subgraph#0
  T#0(serving_default_x:0) shape:[1, 8, 1], type:FLOAT32
  T#1(conv1d/ExpandDims) shape:[4], type:INT32 RO 16 bytes, buffer: 2, data:[1, 1, 8, 1]
  T#2(conv1d/Squeeze) shape:[3], type:INT32 RO 12 bytes, buffer: 3, data:[1, 8, 1]
  T#3(conv1d) shape:[1], type:FLOAT32 RO 4 bytes, buffer: 4, data:[0]
  T#4(conv1d1) shape:[1, 1, 3, 1], type:FLOAT32 RO 12 bytes, buffer: 5, data:[0.812782, -0.248076, -0.0599043]
  T#5(conv1d/ExpandDims1) shape:[1, 1, 8, 1], type:FLOAT32
  T#6(conv1d2) shape:[1, 1, 8, 1], type:FLOAT32
  T#7(PartitionedCall:0) shape:[1, 8, 1], type:FLOAT32
```
The [Netron](https://netron.app) graph shows the dilation parameter is preserved for use by the `CONV2D` kernel:
[image]

The output from TensorFlow, TfLite, and TFLM will now be identical.

The underlying issue within TFLM (`SPACE_TO_BATCH_ND` and `BATCH_TO_SPACE_ND`) for the model generated by Keras is currently being discussed.

ddavis-2015, 2023-11-19:
@apage224 

It turns out you can also get the results you want by using the concrete function method of the `TfLiteConverter` with Keras:

```
# Create example dilated convolutional model
inputs = tf.keras.layers.Input(shape=INPUT_SHAPE)
y = inputs
y = tf.keras.layers.Convolution1D(1, 3, dilation_rate=DILATION_RATE, padding='same')(y)
model = tf.keras.Model(inputs=inputs, outputs=y)
model_func = tf.function(func=model)
cf = model_func.get_concrete_function(
          tf.TensorSpec(shape=(1,)+INPUT_SHAPE, dtype=tf.float32))

# Convert to TF Lite model
converter = tf.lite.TFLiteConverter.from_concrete_functions([cf], model)
tflite_model = converter.convert()
```

apage224, 2023-11-20:
Thanks for looking into this. The proposed workaround does indeed work for me.

closed 2023-11-28 by mergify[bot]

ddavis-2015, 2023-11-28:
PR#[2335](https://github.com/tensorflow/tflite-micro/pull/2335) and PR#[2334](https://github.com/tensorflow/tflite-micro/pull/2334) have been submitted.  24 hours after both PRs have been merged, an automatic update will be pushed to PyPI.

PR #2335 (2023-11-28): SPACE_TO_BATCH_ND: update output tensor shape
@tensorflow/micro

Update the output tensor shape during prepare phase when the computed shape does not match the shape in the flatbuffer.

Added additional tests from TfLite.

See #2319 for additional details.

Resolves [b/313360569](https://issuetracker.google.com/313360569)

bug=fixes #2332 fixes #2319 #1646 #1629 #1231
commit 50433190 (2023-11-28): SPACE_TO_BATCH_ND: update output tensor shape (#2335)

@tensorflow/micro

Update the output tensor shape during prepare phase when the computed shape does not match the shape in the flatbuffer.

Added additional tests from TfLite.

See #2319 for additional details.

Resolves [b/313360569](https://issuetracker.google.com/313360569)

bug=fixes #2332 fixes #2319 #1646 #1629 #1231