tensorflow/tflite-micro #2320: quantization hello_world model got hardfault at AllocateTensors on cortex-m7

ducbxfsoft, 2023-11-20:
quantization hello_world model got hardfault at AllocateTensors on cortex-m7
the example works well with float data model
build TFLM lib command:
make -f tensorflow/lite/micro/tools/make/Makefile BUILD_TYPE=debug TARGET=cortex_m_generic TARGET_ARCH=cortex-m7+fp OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET_TOOLCHAIN_ROOT=/opt/gcc-10.2-arm32-eabi/bin/ microlite

please help to check it, thanks

mansnils, 2023-11-20:
Hi @ducbxfsoft 
Can you provide some more level of detail on how to reproduce this?

ducbxfsoft, 2023-11-21:
hello, I have built/run the hello_world example https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/hello_world_test.cc#L102 on cortex-m7
I got hardfault when run LoadQuantModelAndPerformInference() at https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/hello_world_test.cc#L120 (LoadFloatModelAndPerformInference()  no problem)

mansnils, 2023-11-21:
I am able to run LoadQuantModelAndPerformInference with TARGET_ARCH=cortex-m7+fp, and this is how I run it:
`make -f  tensorflow/lite/micro/tools/make/Makefile  BUILD_TYPE=debug OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m7+fp TOOLCHAIN=gcc  test_hello_world_test`
How do you run it?

ducbxfsoft, 2023-11-22:
I use "make -f tensorflow/lite/micro/tools/make/Makefile BUILD_TYPE=debug TARGET=cortex_m_generic TARGET_ARCH=cortex-m7+fp OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET_TOOLCHAIN_ROOT=custom_toolchan microlite" to build tflite-micro lib and then build hello_world example using that lib (sorry because I can not share the custom_toolchan)
 
But I do the same way to run hello_world example and it works well (float model + quantized model) on cortext-a53, cortext-r52.
But on cortext-m7, only float model works well, quantized model got hardfault at https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/kernels/kernel_util.cc#L320.
 
Can you provide TOOLCHAIN=gcc ? is it public?

mansnils, 2023-11-22:
TOOLCHAIN=gcc is actually default. Yes it is public. So it will use the downloaded third party GCC version.
I think you should first verify that this works:
`make -f  tensorflow/lite/micro/tools/make/Makefile  BUILD_TYPE=debug OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m7+fp test_hello_world_test`
And if that works try with your custom toolchain:
`make -f  tensorflow/lite/micro/tools/make/Makefile  BUILD_TYPE=debug OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m7+fp TARGET_TOOLCHAIN_ROOT=custom_toolchan  test_hello_world_test`

ducbxfsoft, 2023-11-23:
Thank you for responding

I think you are using the different target (cortex_m_corstone_300) so I got hardfault before go to inside my test (note that I can run float model of hello_world example with TARGET=cortex_m_generic, and both of quantized and float model works well on cortext-r52)
Can you explain about cortex_m_corstone_300 ? I am running the example on baremetal environment (no os), is it ok ?
I also using gcc default to build example but got the same issue with my custom_toolchain.

mansnils, 2023-11-23:
`TARGET=cortex_m_corstone_300` was just meant as means to an end here. So if the first one works but not the seconds, it would hint that it was something with your toolchain. And if the seconds one works as well it hints that it is something with your setup, when linking the microlib to your application.
If you want to know more about `TARGET=cortex_m_corstone_300` this is a good starting point: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/cortex_m_corstone_300/README.md

closed 2023-12-25 by github-actions[bot]