tensorflow/tflite-micro #2338: Shape assertion and unintialized memory in Conv2D

cosw0t, 2023-11-30:
I have a simple keras model made of a Conv2D with dilation_rate of 2.

the first issue is in main and the error is reading the shapes in, the assertion here:
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/kernels/internal/types.h
called from here:
https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/kernels/internal/reference/integer_ops/conv.h#L52C4

To reproduce: https://github.com/cosw0t/tflite-micro/tree/bug_report/conv2d_main

It asserts because it has stored input shape: [4, 1, 49, 3] and output shape: [1, 1, 45, 30], but the input shape is wrong, it should be [1, 1, 49, 3]. The shape is correctly reported by Netron as `[1, 1, 49, 3]`.

Second issue:
Following the discussion in #2336 I have rebased onto https://github.com/rascani/tflite-micro/tree/initialize_values that fixes #2337, however it doesn't include #2335 that is instead in main.

however now msan complains about uninitialized memory. The model is the same.\
To reproduce: https://github.com/cosw0t/tflite-micro/tree/bug_report/conv2d_2337

rascani, 2023-11-30:
@ddavis-2015

rascani, 2023-11-30:
Thanks, this example helps. This issue is likely due to #2335, because that reshapes the SpaceToBatch output tensor, which is the input tensor to Conv2d. This was to match TFLite behavior, but I think we likely missed something. I'll probably revert that PR unless there's a simple fix.

ddavis-2015, 2023-11-30:
@cosw0t Please see #2319 for a workaround when converting your Keras model using the TfLiteConverter, as your issue is specific to using convolution with dilation.

cosw0t, 2023-12-01:
> @cosw0t Please see #2319 for a workaround when converting your Keras model using the TfLiteConverter, as your issue is specific to using convolution with dilation.

Apologies, I didn't realize that this was the same issue.
It seems to be the case that converting multiple functions is not supported, so if my model has many layers it's not really an option, right?

thanks

ddavis-2015, 2024-01-03:
PR[#2356](https://github.com/tensorflow/tflite-micro/pull/2356) resolves this issue.

Resolution Verified in Colab.

closed 2024-01-03 by ddavis-2015