tensorflow/tflite-micro #2354: no such package '@tflm_pip_deps_tensorflow_intel//'

Aztec02, 2023-12-12:
Trying to run the hello_world example with bazel ("bazel build tensorflow/lite/micro/examples/hello_world:evaluate")
I get the following error:

ERROR: C:/users/userid/appdata/roaming/spb_data/_bazel_userid/zm73hpx2/external/tflm_pip_deps_tensorflow_cpu/BUILD.bazel:22:11: no such package '@tflm_pip_deps_tensorflow_intel//': The repository '@tflm_pip_deps_tensorflow_intel' could not be resolved: Repository '@tflm_pip_deps_tensorflow_intel' is not defined and referenced by '@tflm_pip_deps_tensorflow_cpu//:pkg'

Environment:
Windows 10
Bazel 6.4.0
Python 3.10.10
Windows cmd prompt

rascani, 2023-12-12:
I wasn't able to reproduce this. Have you modified the python_requirements.txt file? Perhaps just a transient pypi download issue?

Aztec02, 2023-12-12:
I just tried that on another Windows 10 machine with Bazel 6.4.0 and got the exact same error.  I have not modified the requirements. I downloaded the bazel-6.4.0-windows-x86_64.exe, renamed it to bazel and added to PATH. Installed the https://aka.ms/vs/17/release/vc_redist.x64.exe Microsoft Visual C++ Redistributable. Cloned the tflite-micro repo. Open the cmd prompt. Typed bazel build tensorflow/lite/micro/examples/hello_world:evaluate and got the exact same error.

I also tried installing the python requirements and running the evaluate.py script manually but there are other package problems there. For example the script imports matplotlib but matplotlib is not in the requirements, etc

Aztec02, 2023-12-15:
Solved this issue by modifying the requirements:
In python_requirements.txt changed tensorflow-io-gcs-filesystem==0.32.0  to version 0.31.0 because there is no wheel for windows and version 32 at pypi.
Also added 
tensorflow-intel==2.12.0 \
    --hash=sha256:816a6b9018d1ae0defe94508aab2512228fb6a54ab51f5f4f025c857cfe5c7e5
    # via tensorflow-cpu 
  as this is a dependency for tensorflow-cpu.
  I build the hello_world:train target successfully.
  But then when I try to build hello_world:evaluate target I get a new error "ERROR: An error occurred during the fetch of repository 'local_config_python':".
  Correct me if I am wrong but this repo needs many modifications to work on windows.

CannoChen, 2023-12-16:
I run into a similar problem as you, and there is no good solution yet. 
I have tried to use Linux System to solve this problem, but I failed...I suspect the official documentation is out of date！！！！
-----
If you have any problemes, Please update in here! I'm sure this will help a lot of people with similar problems.

CannoChen, 2023-12-16:
I sloved this problem. Let me tell you how to slove it!!!
First, you must create a new virtual env by using conda:
`conda create -n myenv python=3.10.0 ` 
It  is import that the version of python must be 3.10!!!!! If your python version is not 3.10, then it will definitely fail!！

secondly, you must install bazel 7.0. 

Finally, don't install any packages, just run the command:
bazel build tensorflow/lite/micro/examples/micro_speech:audio_preprocessor
(you can instead this command for yours~)

Now,  everything is ok!!!

Aztec02, 2023-12-16:
Tried in Ubuntu jellyfish and everything builds fine.

CannoChen, 2023-12-16:
Hi,This is CannoChen.I have received your mail.And then, I will reply to you later.Thank you!

liuchangzong, 2023-12-26:
> Solved this issue by modifying the requirements: In python_requirements.txt changed tensorflow-io-gcs-filesystem==0.32.0 to version 0.31.0 because there is no wheel for windows and version 32 at [...]

That's really work!

liuchangzong, 2023-12-26:
> That's really work!

Well, unfortunately, I got another issue when I was trying to "bazel-bin" the .exe file. Something like this:
`E:\tinyML\tflite-micro>bazel-bin\tensorflow\lite\micro\examples\hello_world\train --save_tf_model --save_dir=\tmp\model_created\
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "E:\tinyML\tflite-micro\bazel-bin\tensorflow\lite\micro\examples\hello_world\train.zip\__main__.py", line 559, in <module>
  File "E:\tinyML\tflite-micro\bazel-bin\tensorflow\lite\micro\examples\hello_world\train.zip\__main__.py", line 454, in Main
  File "E:\tinyML\tflite-micro\bazel-bin\tensorflow\lite\micro\examples\hello_world\train.zip\__main__.py", line 204, in CreateModuleSpace
  File "E:\tinyML\tflite-micro\bazel-bin\tensorflow\lite\micro\examples\hello_world\train.zip\__main__.py", line 190, in ExtractZip
  File "D:\Python\Python311\Lib\zipfile.py", line 1660, in extract
    return self._extract_member(member, path, pwd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Python\Python311\Lib\zipfile.py", line 1731, in _extract_member
    open(targetpath, "wb") as target:
    ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\MyName\\AppData\\Local\\Temp\\Bazel.runfiles_93cfifk4\\runfiles\\tflm_pip_deps_tensorflow_intel\\site-packages\\tensorflow\\include\\external\\llvm-project\\mlir\\_virtual_includes\\AffineMemoryOpInterfacesIncGen\\mlir\\Dialect\\Affine\\IR\\AffineMemoryOpInterfaces.cpp.inc'`

closed 2024-04-26 by ddavis-2015