tensorflow/tflite-micro #2359: rfft2d Operation

ericheim, 2023-12-15:
### Description
When creating a tf.Module using a rfft operation from ```tensorflow.signal``` the resulting tflite model includes the rfft2d operation and expands the tensor to 2D. To run the exported model in tfmicro an implementation of rfft2d is required.

### How to reproduce:
Example code:
```
import tensorflow as tf

class MyRfft(tf.Module):
    def __init__(self, frame_length, **kwargs):
        super(MyRfft, self).__init__(**kwargs)
        self._frame_length = frame_length

    @tf.function
    def inference_fn(self, inputs):
        # create spectogram
        spec = tf.signal.rfft(inputs)
        spec = tf.abs(spec)
        spec = spec[..., tf.newaxis]
        return spec

model = MyRfft(255)
concrete_function = model.inference_fn.get_concrete_function(
            tf.TensorSpec(shape=[256], dtype=tf.float32, name='input'))

converter = tf.lite.TFLiteConverter.from_concrete_functions([concrete_function])
tflite_model = converter.convert()
tflite_model_path = "rfft.tflite"
with open(tflite_model_path, 'wb') as f:
  f.write(tflite_model)
```

### Solution
Implement rfft2d (or maybe fix in tensorflow repo?)

#### Graph of rfft.tflite:
[image]

ddavis-2015, 2023-12-22:
@ericheim 

There is good news and bad news regarding "signal" libraries.

The bad: You will not be able to use `tf.signal` with TFLM.  The TFLM runtime does not support the `Rfft2d` and `ComplexAbs` operations.  Also, TFLM does not have support for a complex datatype.

The good: TFLM has its own signal library.  The library includes `SignalRfft` and `SignalEnergy`.  The `SignalRfft` operation would require int16 inputs and outputs pairs of int16 values representing the real/imaginary portions of the complex values.  The `SignalEnergy` operation requires int16 pairs (complex values), and outputs uint32 values.  I believe the output values are computed as: (real * real) + (imaginary * imaginary).

The TFLM signal library is implemented as a set of TensorFlow Python plugins, and as TFLM kernels.  This allows for use in Python scripts when using TensorFlow eager or graph execution, as well as TfLiteConverter -> MicroInterpreter model inference from within Python scripts.

The TFLM signal library is not yet available as part of the PyPI download.

Some documentation for the TFLM signal operations is available within the source code here: [plugin ops](https://github.com/tensorflow/tflite-micro/tree/main/signal/tensorflow_core/ops)

An example using the TFLM signal ops can be found here: [micro_speech example](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech)
Of interest here would be the audio_preprocessor.py and BUILD scripts.  You will need to clone the repo, and create a new example application and Bazel BUILD script (based on the micro_speech example).

ericheim, 2024-01-02:
@ddavis-2015 Thanks for the explanation. Right now the signal ops used in the [micro_speech example](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech) are not working out of the box with the tflite_micro python package. I have a pending PR for this #2358

closed 2024-02-02 by github-actions[bot]