tensorflow/tflite-micro #2369: Building tflite-micro gives error fixedpoint.fixedpoint.h:No such file directory

JasmineKaur-eaton, 2023-12-25:
On building tflite-micro gives error fixedpoint.fixedpoint.h:No such file directory error is found. 
I have downloaded the tflite-micro_main from the GIT.

while including the #include "tflite-micro/tensorflow/lite/micro/micro_mutable_op_resolver.h" in the file tflite-micro\tensorflow\lite\kernels\internal\common.h this error is found at this path.

can you suggest me how to resolve this error?  

let me know if any other information is required.

rascani, 2024-01-02:
`fixedpoint/fixedpoint.h` comes from the external dependency gemmlowp. This dependency is downloaded as part of the build (regardless of Bazel or make). You're likely missing that dependency (as well as others).

closed 2024-01-02 by rascani

JasmineKaur-eaton, 2024-01-02:
Can you please share how can I download the missing dependency?

rascani, 2024-01-02:
They're automatically downloaded by the bazel or make build.

If you'd like to just trigger the dependencies download, you can use this make target:

```
make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads
```

JasmineKaur-eaton, 2024-01-02:
On executing the above cmd, following error is thrown- 
process_begin: CreateProcess(NULL, uname -m, ...) failed.
tensorflow/lite/micro/tools/make/Makefile:47: pipe: No error
-m was unexpected at this time.
FIND: Parameter format not correct
FIND: Parameter format not correct
File not found - "*.c"
Access denied - TENSORFLOW/LITE
File not found - -TYPE
File not found - D
File not found - \(
File not found - -PATH
Access denied - TENSORFLOW/LITE/EXPERIMENTAL
File not found - -O
File not found - -PATH
Access denied - TENSORFLOW/LITE/MICRO
File not found - \)
File not found - -PRUNE
File not found - -FALSE
File not found - -O
File not found - -NAME
The syntax of the command is incorrect.

Can you suggest what can be done fix it?

rascani, 2024-01-02:
Ah, sorry I missed that you were using Windows. The make build doesn't support Windows, since it uses linux utilities. You might have better luck with the bazel build.

JasmineKaur-eaton, 2024-01-02:
Can you point me to the cmds using bazel?