tensorflow/tflite-micro #2380: atal error: tensorflow/lite/micro/micro_mutable_op_resolver.h: No such file or directory    19 | #include <tensorflow/lite/micro/micro_mutable_op_resolver.h>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ compilation terminated.

arunrajvakathanam, 2024-01-06:
try to run hello_world program in from the folder "zephyrproject/zephyr/samples/modules/tflite-micro/hello_world" i got the error:

ubuntu22:~/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world$ west build -b qemu_x86 .

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world
-- CMake version: 3.28.1
-- Found Python3: /home/arunraj/zephyrproject/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/arunraj/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/arunraj/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: qemu_x86
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.4 (/home/arunraj/zephyr-sdk-0.16.4)
-- Found toolchain: zephyr 0.16.4 (/home/arunraj/zephyr-sdk-0.16.4)
-- Found Dtc: /home/arunraj/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/arunraj/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86.dts
-- Generated zephyr.dts: /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/dts.cmake

warning: TENSORFLOW_LITE_MICRO (defined at modules/tflite-micro/Kconfig:8) was assigned the value
'y' but got the value 'n'. Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_TENSORFLOW_LITE_MICRO and/or look up
TENSORFLOW_LITE_MICRO in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/arunraj/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/arunraj/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86_defconfig'
Merged configuration '/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/prj.conf'
Configuration saved to '/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/arunraj/zephyr-sdk-0.16.4/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/arunraj/zephyr-sdk-0.16.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CMake Warning at /home/arunraj/zephyrproject/zephyr/subsys/random/CMakeLists.txt:12 (message):
  

      Warning: CONFIG_TIMER_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

-- Using ccache: /usr/bin/ccache
-- Configuring done (23.2s)
-- Generating done (0.1s)
-- Build files have been written to: /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build
-- west build: building application
[1/119] Preparing syscall dependency handling

[3/119] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/home/arunraj/zephyrproject/zephyr), build: zephyr-v3.5.0-3680-g7d3eee7b272d
[13/119] Building CXX object CMakeFiles/app.dir/src/main_functions.cpp.obj
FAILED: CMakeFiles/app.dir/src/main_functions.cpp.obj 
ccache /home/arunraj/zephyr-sdk-0.16.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++ -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/arunraj/zephyrproject/zephyr/include -I/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/include/generated -I/home/arunraj/zephyrproject/zephyr/soc/x86/ia32 -isystem /home/arunraj/zephyrproject/zephyr/lib/libc/common/include -isystem /home/arunraj/zephyrproject/zephyr/lib/cpp/minimal/include -m32  -fno-strict-aliasing -Os -fcheck-new -std=c++17 -Wno-register -fno-exceptions -fno-rtti -imacros /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -m32 -msoft-float -Wa,--divide --sysroot=/home/arunraj/zephyr-sdk-0.16.4/x86_64-zephyr-elf/x86_64-zephyr-elf -imacros /home/arunraj/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/arunraj/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/arunraj/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mpreferred-stack-boundary=2 -mno-mmx -mno-sse --specs=picolibc.specs -nostdinc++ -fno-threadsafe-statics -MD -MT CMakeFiles/app.dir/src/main_functions.cpp.obj -MF CMakeFiles/app.dir/src/main_functions.cpp.obj.d -o CMakeFiles/app.dir/src/main_functions.cpp.obj -c /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/main_functions.cpp
/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/main_functions.cpp:19:10: fatal error: tensorflow/lite/micro/micro_mutable_op_resolver.h: No such file or directory
   19 | #include <tensorflow/lite/micro/micro_mutable_op_resolver.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[14/119] Building CXX object CMakeFiles/app.dir/src/output_handler.cpp.obj
FAILED: CMakeFiles/app.dir/src/output_handler.cpp.obj 
ccache /home/arunraj/zephyr-sdk-0.16.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++ -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/arunraj/zephyrproject/zephyr/include -I/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/include/generated -I/home/arunraj/zephyrproject/zephyr/soc/x86/ia32 -isystem /home/arunraj/zephyrproject/zephyr/lib/libc/common/include -isystem /home/arunraj/zephyrproject/zephyr/lib/cpp/minimal/include -m32  -fno-strict-aliasing -Os -fcheck-new -std=c++17 -Wno-register -fno-exceptions -fno-rtti -imacros /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -m32 -msoft-float -Wa,--divide --sysroot=/home/arunraj/zephyr-sdk-0.16.4/x86_64-zephyr-elf/x86_64-zephyr-elf -imacros /home/arunraj/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/arunraj/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/arunraj/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mpreferred-stack-boundary=2 -mno-mmx -mno-sse --specs=picolibc.specs -nostdinc++ -fno-threadsafe-statics -MD -MT CMakeFiles/app.dir/src/output_handler.cpp.obj -MF CMakeFiles/app.dir/src/output_handler.cpp.obj.d -o CMakeFiles/app.dir/src/output_handler.cpp.obj -c /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/output_handler.cpp
In file included from /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/src/output_handler.cpp:17:
[... cut ...]
   20 | #include <tensorflow/lite/c/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[15/119] Building CXX object CMakeFiles/app.dir/src/model.cpp.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/arunraj/zephyrproject/zephyr/samples/modules/tflite-micro/hello_world/build

my system configuration is

ubuntu :22.04
CMake version: 3.28.1
Python3:3.10.12
Zephyr version: 3.5.99
west version "1.2.0"
Board: qemu_x86
zephyr-sdk:zephyr 0.16.4

closed 2024-02-05 by github-actions[bot]

alesimattia, 2024-05-30:
Hell, I have the same error on Zephyr SDK 0.16.5-1
`/Users/alesimattia/zephyrproject/zephyr/samples/modules/tflite-micro/magic_wand/src/output_handler.hpp:20:10: fatal error: tensorflow/lite/c/common.h: No such file or directory`

I noticed that in the CMake.lists at the path 
>zephyrproject/zephyr/modules/tflite-micro
actually have this line o _${TENSORFLOW_LITE_MICRO_DIR}/tensorflow/lite/core/c/common.cc_

arunrajvakathanam, 2024-05-30:
The latest version of Zephyr SDK not include the tensorflow-lite

alesimattia, 2024-05-30:
Well it would've been nice to write down in the related [documentation page](https://docs.zephyrproject.org/latest/samples/modules/tflite-micro/hello_world/README.html
) ehehe

As found in [another discussion](url) you have to properly edit the _west_ configuration as above:
```
west config manifest.project-filter -- +tflite-micro
west config manifest.group-filter -- +optional
west update
```

amanem1, 2025-03-15:
When i tried to run the  "zephyrproject\zephyr\samples\modules\tflite-micro\hello_world" , build failed with 
1. ../tf_common.h not found .  then i performed  west update -> after west update
2. .No such file or directory 19 | #include <tensorflow/lite/micro/micro_mutable_op_resolver.h>

then tried running 

> west config manifest.project-filter -- +tflite-micro
west config manifest.group-filter -- +optional
west update

After then if i rebuild, it works .