tensorflow/tflite-micro #2404: hello world:evaluate bazel build failure caused by FLATBUFFERS_ASSERT

c-z-c-2, 2024-01-25:
### When running 
`bazel build tensorflow/lite/micro/examples/hello_world:evaluate`

### I get the following error and haven't been able to resolve it.

INFO: Analyzed target //tensorflow/lite/micro/examples/hello_world:evaluate (0 packages loaded, 0 targets configured).
ERROR: /home/x/tflite-micro/tensorflow/lite/micro/kernels/BUILD:194:23: Compiling tensorflow/lite/micro/kernels/var_handle.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target //tensorflow/lite/micro/kernels:micro_ops) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 43 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/base.h:20,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/array.h:23,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flatbuffers.h:24,
                 from ./tensorflow/lite/micro/flatbuffer_utils.h:19,
                 from ./tensorflow/lite/micro/micro_allocator.h:23,
                 from ./tensorflow/lite/micro/micro_resource_variable.h:22,
                 from ./tensorflow/lite/micro/micro_graph.h:21,
                 from ./tensorflow/lite/micro/micro_context.h:20,
                 from ./tensorflow/lite/micro/kernels/kernel_util.h:26,
                 from tensorflow/lite/micro/kernels/var_handle.cc:24:
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h: In constructor 'flexbuffers::Verifier::Verifier(const uint8_t*, size_t, std::vector<unsigned char>*, bool, size_t)':
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:30: error: comparison of integer expressions of different signedness: 'size_t' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:5: note: in expansion of macro 'FLATBUFFERS_ASSERT'
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
      |     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
ERROR: /home/x/tflite-micro/tensorflow/lite/micro/BUILD:73:11: Compiling tensorflow/lite/micro/micro_interpreter_context.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target //tensorflow/lite/micro:micro_interpreter_context) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 31 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/base.h:20,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/array.h:23,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flatbuffers.h:24,
                 from ./tensorflow/lite/micro/flatbuffer_utils.h:19,
                 from ./tensorflow/lite/micro/micro_allocator.h:23,
                 from ./tensorflow/lite/micro/micro_interpreter_context.h:20,
                 from tensorflow/lite/micro/micro_interpreter_context.cc:16:
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h: In constructor 'flexbuffers::Verifier::Verifier(const uint8_t*, size_t, std::vector<unsigned char>*, bool, size_t)':
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:30: error: comparison of integer expressions of different signedness: 'size_t' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:5: note: in expansion of macro 'FLATBUFFERS_ASSERT'
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
      |     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
ERROR: /home/x/tflite-micro/tensorflow/lite/micro/kernels/BUILD:111:11: Compiling tensorflow/lite/micro/kernels/kernel_util.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target //tensorflow/lite/micro/kernels:kernel_util) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 35 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/base.h:20,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/array.h:23,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flatbuffers.h:24,
                 from ./tensorflow/lite/micro/flatbuffer_utils.h:19,
                 from ./tensorflow/lite/micro/micro_allocator.h:23,
                 from ./tensorflow/lite/micro/micro_resource_variable.h:22,
                 from ./tensorflow/lite/micro/micro_graph.h:21,
                 from ./tensorflow/lite/micro/micro_context.h:20,
                 from ./tensorflow/lite/micro/kernels/kernel_util.h:26,
                 from tensorflow/lite/micro/kernels/kernel_util.cc:16:
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h: In constructor 'flexbuffers::Verifier::Verifier(const uint8_t*, size_t, std::vector<unsigned char>*, bool, size_t)':
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:30: error: comparison of integer expressions of different signedness: 'size_t' {aka 'long unsigned int'} and 'int' [-Werror=sign-compare]
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:5: note: in expansion of macro 'FLATBUFFERS_ASSERT'
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
      |     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
ERROR: /home/x/tflite-micro/tensorflow/lite/micro/BUILD:135:11: Compiling tensorflow/lite/micro/micro_interpreter_graph.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target //tensorflow/lite/micro:micro_interpreter_graph) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 31 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
In file included from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/base.h:20,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/array.h:23,
                 from bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flatbuffers.h:24,
                 from ./tensorflow/lite/micro/flatbuffer_utils.h:19,
[... 103 lines cut ...]
cc1plus: all warnings being treated as errors
ERROR: /home/x/tflite-micro/tensorflow/lite/micro/BUILD:56:11: Compiling tensorflow/lite/micro/micro_context.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target //tensorflow/lite/micro:micro_context) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++14' -MD -MF ... (remaining 31 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
[... cut ...]
bazel-out/k8-fastbuild/bin/external/flatbuffers/_virtual_includes/runtime_cc/flatbuffers/flexbuffers.h:1672:5: note: in expansion of macro 'FLATBUFFERS_ASSERT'
 1672 |     FLATBUFFERS_ASSERT(size_ < FLATBUFFERS_MAX_BUFFER_SIZE);
      |     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
Target //tensorflow/lite/micro/examples/hello_world:evaluate failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 57.469s, Critical Path: 53.18s
INFO: 24 processes: 24 internal.
ERROR: Build did NOT complete successfully

closed 2024-02-25 by github-actions[bot]

kyleaitgroup, 2024-03-13:
Hello @c-z-c-2 is the problem resolved? I've encountered the same issue.

kisielk, 2024-04-16:
@kyleaitgroup try adding `--copt='-Wno-error=sign-compare' ` to the bazel invocation