tensorflow/tflite-micro #2448: Error in fail: whl_library tflm_pip_deps_numpy failed

bluepikaqiu, 2024-02-07:
$:bazel run tensorflow/lite/micro/examples/hello_world:hello_world_test

ERROR: An error occurred during the fetch of repository 'tflm_pip_deps_numpy':                                                                
   Traceback (most recent call last):                                  
        File "/home/user1/.cache/bazel/_bazel_user1/59a1cd38bbbb3e5e59482d3228ca5ed1/external/rules_python/python/pip_install/pip_repository.bzl", line 691, column 13, in _whl_library_impl                           
                fail("whl_library %s failed: %s (%s) error code: '%s'" % (rctx.attr.name, result.stdout, result.stderr, result.return_code))  
Error in fail: whl_library tflm_pip_deps_numpy failed:  (Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 174, in _run_module_as_main 
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)   
  File "/usr/lib/python3.5/runpy.py", line 144, in _get_module_details 
    code = loader.get_code(mod_name)
  File "<frozen importlib._bootstrap_external>", line 767, in get_code 
  File "<frozen importlib._bootstrap_external>", line 727, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/user1/.cache/bazel/_bazel_user1/59a1cd38bbbb3e5e59482d3228ca5ed1/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 336
    entry_point_without_py = f"{name[:-3]}_py" if name.endswith(".py") else name
                                             ^
SyntaxError: invalid syntax
) error code: '1'
ERROR: /home/user1/tflite-micro/WORKSPACE:55:13: fetching whl_library rule //external:tflm_pip_deps_numpy: Traceback (most recent call last):  
        File "/home/user1/.cache/bazel/_bazel_user1/59a1cd38bbbb3e5e59482d3228ca5ed1/external/rules_python/python/pip_install/pip_repository.bzl", line 691, column 13, in _whl_library_impl                           
                fail("whl_library %s failed: %s (%s) error code: '%s'" % (rctx.attr.name, result.stdout, result.stderr, result.return_code))  
Error in fail: whl_library tflm_pip_deps_numpy failed:  (Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 174, in _run_module_as_main 
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)   
  File "/usr/lib/python3.5/runpy.py", line 144, in _get_module_details 
    code = loader.get_code(mod_name)
  File "<frozen importlib._bootstrap_external>", line 767, in get_code 
  File "<frozen importlib._bootstrap_external>", line 727, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/user1/.cache/bazel/_bazel_user1/59a1cd38bbbb3e5e59482d3228ca5ed1/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 336
    entry_point_without_py = f"{name[:-3]}_py" if name.endswith(".py") else name
                                             ^
SyntaxError: invalid syntax
) error code: '1'
INFO: Repository gemmlowp instantiated at:
  /home/user1/tflite-micro/WORKSPACE:20:10: in <toplevel>
  /home/user1/tflite-micro/tensorflow/workspace.bzl:63:20: in workspace 
  /home/user1/tflite-micro/tensorflow/workspace.bzl:30:20: in tf_repositories
  /home/user1/tflite-micro/third_party/repo.bzl:90:21: in tf_http_archive
Repository rule _tf_http_archive defined at:
  /home/user1/tflite-micro/third_party/repo.bzl:65:35: in <toplevel>    
INFO: Repository ruy instantiated at:
  /home/user1/tflite-micro/WORKSPACE:20:10: in <toplevel>
  /home/user1/tflite-micro/tensorflow/workspace.bzl:10:8: in initialize_third_party
  /home/user1/tflite-micro/third_party/ruy/workspace.bzl:6:20: in repo
  /home/user1/tflite-micro/third_party/repo.bzl:90:21: in tf_http_archive
Repository rule _tf_http_archive defined at:
  /home/user1/tflite-micro/third_party/repo.bzl:65:35: in <toplevel>
INFO: Repository kissfft instantiated at:
  /home/user1/tflite-micro/WORKSPACE:20:10: in <toplevel>
  /home/user1/tflite-micro/tensorflow/workspace.bzl:63:20: in workspace
  /home/user1/tflite-micro/tensorflow/workspace.bzl:60:27: in tf_repositories
  /home/user1/tflite-micro/tensorflow/workspace.bzl:9:12: in initialize_third_party
  /home/user1/tflite-micro/third_party/kissfft/workspace.bzl:6:20: in repo
  /home/user1/tflite-micro/third_party/repo.bzl:90:21: in tf_http_archive
Repository rule _tf_http_archive defined at:
  /home/user1/tflite-micro/third_party/repo.bzl:65:35: in <toplevel>
ERROR: no such package '@@tflm_pip_deps_numpy//': whl_library tflm_pip_deps_numpy failed:  (Traceback (most recent call last):                                                                                      
  File "/usr/lib/python3.5/runpy.py", line 174, in _run_module_as_main                                    
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)                                      
  File "/usr/lib/python3.5/runpy.py", line 144, in _get_module_details                                    
    code = loader.get_code(mod_name)                                                                      
  File "<frozen importlib._bootstrap_external>", line 767, in get_code                                    
  File "<frozen importlib._bootstrap_external>", line 727, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/user1/.cache/bazel/_bazel_user1/59a1cd38bbbb3e5e59482d3228ca5ed1/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 336
    entry_point_without_py = f"{name[:-3]}_py" if name.endswith(".py") else name
                                             ^
SyntaxError: invalid syntax
) error code: '1'
ERROR: /home/user1/tflite-micro/tensorflow/lite/micro/tools/BUILD:40:10: //tensorflow/lite/micro/tools:generate_cc_arrays depends on @@tflm_pip_deps_numpy//:pkg in repository @@tflm_pip_deps_numpy which failed to fetch. no such package '@@tflm_pip_deps_numpy//': whl_library tflm_pip_deps_numpy failed:  (Traceback (most recent call last):                                                                                      
  File "/usr/lib/python3.5/runpy.py", line 174, in _run_module_as_main                                    
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)                                      
  File "/usr/lib/python3.5/runpy.py", line 144, in _get_module_details
    code = loader.get_code(mod_name)
  File "<frozen importlib._bootstrap_external>", line 767, in get_code
  File "<frozen importlib._bootstrap_external>", line 727, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/user1/.cache/bazel/_bazel_user1/59a1cd38bbbb3e5e59482d3228ca5ed1/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 336
    entry_point_without_py = f"{name[:-3]}_py" if name.endswith(".py") else name
                                             ^
SyntaxError: invalid syntax
) error code: '1'
ERROR: Analysis of target '//tensorflow/lite/micro/examples/hello_world:hello_world_test' failed; build aborted: Analysis failed
INFO: Elapsed time: 5.892s, Critical Path: 0.07s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
ERROR: Build failed. Not running target
FAILED:
    Fetching https://storage.googleapis.com/...40caa49e570745.zip; 4s
    Fetching https://storage.googleapis.com/...193629d4bd42de4a59.zip
    Fetching https://github.com/.../archive/refs/tags/v130.zip

   Howt to solve this problem?

SL-RU, 2024-03-02:
I have the same problem

rascani, 2024-03-04:
I haven't encountered this before. The error seems to be indicating a problem with the wheel_installer, which certainly isn't a part of our codebase. You may want to try clearing the bazel cache. `bazel clean --expunge`.

594511153, 2024-03-19:
you should install JDK

closed 2024-04-18 by github-actions[bot]

Uwaang, 2024-05-08:
i have a same issue