tensorflow/tflite-micro #2461: Sound (not speech) recognition on esp32-s3-eye v2.2

gamename, 2024-02-20:
Hi,

I am trying to do sound recognition (specifically, a cat meowing).  My platform is the esp32-s3-eye. Is this the best library to use?  I already have the TinyML model.

Thanks,
-T

ddavis-2015, 2024-02-23:
@gamename Hello, and thank you for your interest in TensorFlow Lite for Microcontrollers (TFLM).  You have several approaches you can use for your esp32-s3 platform:

1. Integrate with the existing TFLM repository.  After cloning the repo, you could add the appropriate esp32-s3 support in these directories:
    - tensorflow/lite/micro/tools/make/targets
        - your esp32-s3 specific makefile parameters
    - tensorflow/lite/micro/
        - your esp32-s3 specific debug_log.cc and system_setup.cc (examples such as the `ceva` and `chre` directories
2. Create your own build environment using CMake and the `project_generation` scripts in:
    - tensorflow/lite/micro/tools/project_generation
        - use the `create_tflm_tree` script with an Xtensa target architecture (P6?)
        - use the `Makefile` according to the instructions inside

This approach will probably require quite a bit of work to get to compile correctly.
3. Arduino.  The esp32-s3 is supported by the Arduino IDE.  You can clone the [TFLM Arduino repo](https://github.com/tensorflow/tflite-micro-arduino-examples) and then follow the instructions [here](https://github.com/tensorflow/tflite-micro/issues/2291#issuecomment-1789452699).  Be sure to substitute the correct esp32-s3 flags for the Adafruit ones. This may allow you to compile the TFLM library and examples.  Note that the Arduino TFLM repo. is from March 2023, and is not up to date with the main TFLM repo.

This issue will now be closed.  Please feel free to reopen the issue should you have additional comments.

closed 2024-02-23 by ddavis-2015

gamename, 2024-02-23:
@ddavis-2015 
> This issue will now be closed. Please feel free to reopen the issue should you have additional comments.
THank you!

ddavis-2015, 2024-02-26:
@gamename 

Suggesting the TFLM Arduino repo. was not correct.  The TFLM Arduino repo. is specific to ARM Cortex processors.

However, Espressif has their own TFLM repository which they keep synched with the official Google TFLM repo.  The Espressif TFLM code supports the esp32-s3 and can be found [here](https://github.com/espressif/esp-tflite-micro).

gamename, 2024-02-26:
> @gamename
>
> Suggesting the TFLM Arduino repo. was not correct. The TFLM Arduino repo. is specific to ARM Cortex processors.
>
> However, Espressif has their own TFLM repository which they keep synched [...]

Thanks!