tensorflow/tflite-micro #2474: Figuring out the *arena size* and required *operations* (before deploying with TFLM)

Black3rror, 2024-02-25:
I have a couple of questions that I would like to ask at once since they are related.

## Finding the arena size
First, I want to check if TFLite or TFLM provides any tool for specifying (or at least estimating) the arena size required for the execution of the network on the end device. For example, a Python script/function that we'll pass the TFLite model to it. Then by analyzing the model, it will figure out the required arena size.

I'm aware that we can do such a thing with `MicroProfiler` (although I did not succeed in doing so, but anyway), still, this analysis is done once we upload the model to the target microcontroller. What I'm asking is an automated way of specifying the arena size while we are developing the Python code and creating C++ files in our Python script.

To make it more concrete, let me exemplify Ekkono. In Ekkono, one can find the required arena size by `arena_size = Ekkono.primer.ModelInspector.crystal_dynamic_memory_usage(model)` and use it when generating/writing the C code that should be uploaded to the board.
Based on theories at least, I think it should be possible to do a similar thing in TFLite or TFLM. **Is there already any tool/code (or any attempt) that can do that?**

Based on the tensorflow model, a naive solution that I'm using is to consider the addition of the memory space required by feature maps of two consecutive layers as the main element using this arena space and just adding 1000 additional bytes to the arena_size for safety. It's worth mentioning that in-place operations (activation functions, for example) should not be considered as a new layer since they don't require additional memory. The code looks something like this:
```Python
def _estimate_arena_size(model):
        def _mul_dims(dims):
            output = 1
            for dim in dims:
                output *= dim
            return output
        arena_size = 0
        layer_1_size = _mul_dims(model.layers[0].input_shape[1:])

        for layer in model.layers:
            if isinstance(layer, tf.keras.layers.Conv2D):
                layer_2_size = _mul_dims(layer.output_shape[1:])
            elif isinstance(layer, tf.keras.layers.BatchNormalization):
                continue
            elif isinstance(layer, tf.keras.layers.Activation):
                continue
            elif isinstance(layer, tf.keras.layers.Dropout):
                continue
            elif isinstance(layer, tf.keras.layers.MaxPooling2D):
                layer_2_size = _mul_dims(layer.output_shape[1:])
            elif isinstance(layer, tf.keras.layers.GlobalAveragePooling2D):
                layer_2_size = _mul_dims(layer.output_shape[1:])
            elif isinstance(layer, tf.keras.layers.Flatten):
                continue
            elif isinstance(layer, tf.keras.layers.Dense):
                layer_2_size = _mul_dims(layer.output_shape[1:])
            else:
                raise ValueError("Unknown layer type: {}".format(layer))

            if layer_1_size + layer_2_size > arena_size:
                arena_size = layer_1_size + layer_2_size
            layer_1_size = layer_2_size

        arena_size = arena_size * 4     # 4 bytes for each float32
        arena_size  += 1000
        return arena_size
```
This approach failed in many cases that I've tested (the required arena size was more than my estimation), which leads me to my second question: **Is my assumption that the arena space will mainly be occupied by feature maps of consecutive layers false? Is there anything else that I should consider?**

## Finding the required operators
Using TFLM we need to pull in different operations. For example, `status = resolver.AddConv2D();`. It should definitely be possible to figure out which operations should be included by looking at the tflite model. So, my third question is: **Is there already any code that does this for us? (or at least any tries?)**

My last question is about the registration of different operations. In some examples available in TFLM, like [hello_world](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/hello_world_test.cc) or [micro_speech](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/micro_speech/micro_speech_test.cc), different operations are added simply, for example, `op_resolver.AddFullyConnected()` or `AddConv2D()`, even for full_int quantized models. While in some other examples, like [person_detection](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/person_detection/person_detection_test.cc), a special registration of them is added, for example, `AddConv2D(tflite::Register_CONV_2D_INT8())`.
**What is the difference between these two ways of registration?** I only used the simpler approach (empty argument) and it just worked well, even for full_int quantized models. So, when should I follow the second approach, and why?

mansnils, 2024-03-22:
A tool for finding arena size would indeed be useful. For finding required operators there is this tool: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/gen_micro_mutable_op_resolver/README.md

For your last question, the latter `AddConv2D(tflite::Register_CONV_2D_INT8())` is a way to only register the int8 kernel. This can be useful for decreasing the memory footprint if you know only int8 model(s) will be used for example. What type of specific quantization is supported depends on what type of kernel implementation it is, see for example: https://github.com/tensorflow/tflite-micro/blob/15092dfdba4007dc7475cdf56f0dfa72ac280355/tensorflow/lite/micro/kernels/fully_connected.h#L79

rascani, 2024-03-22:
We don't have a python based tool for determining arena size, but we do have a C++ one. The arena is primarily comprised of tensors, organized in such a way to maximize re-use and minimize arena size. This would be fairly easy to estimate via Python. However, there are additional allocations from each operator for various data structures. These are normally comprised of various pieces of pre-calculated data that helps keep inference from repeating calculations. The tricky bit for estimating arena size comes from these data structures. They can vary for each operator, on each target hardware. Ie, the ARM cmsis_nn version of Conv2D will allocate different data structures than the Xtensa nnlib version of Conv2D. 

Our recommended approach would be to use the [benchmark utility](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/tools/benchmarking) with your model and a simulator/emulator of your target hardware. It will output various profiling data and breaks down both persistent and non-persistent tensor arena use.

Black3rror, 2024-03-23:
@mansnils @rascani 
I think you answered all my questions.
Thanks a lot!

closed 2024-03-23 by Black3rror