tensorflow/tflite-micro #2485: Link Errors Building `tflite_micro` Python Package

gamename, 2024-03-02:
Hi,

Per the instructions [here](https://github.com/tensorflow/tflite-micro/blob/main/python/tflite_micro/README.md), I tried building `tflite_micro` and got these errors (below).

My env:
  MacOS 14.2.1 (23C71)
  Chip Apple M1 Max
  XCode Version 15.2 (15C500b)
  Python 3.11.7
  bazel 7.0.2-homebrew

Command:
  `bazel build //python/tflite_micro:whl.dist`

Output:
```
Starting local Bazel server and connecting to it...
INFO: Analyzed target //python/tflite_micro:whl.dist (114 packages loaded, 14100 targets configured).
INFO: From Linking external/ruy/ruy/profiler/libinstrumentation.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: archive library: bazel-out/darwin_arm64-fastbuild/bin/external/ruy/ruy/profiler/libinstrumentation.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking tensorflow/lite/kernels/internal/libtensor_utils_no_eigen.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning renaming duplicate member name 'portable_tensor_utils.o' from 'bazel-out/darwin_arm64-fastbuild/bin/tensorflow/lite/kernels/internal/_objs/tensor_utils_no_eigen/0/portable_tensor_utils.o(portable_tensor_utils.o)' and 'bazel-out/darwin_arm64-fastbuild/bin/tensorflow/lite/kernels/internal/_objs/tensor_utils_no_eigen/1/portable_tensor_utils.o(portable_tensor_utils.o)'
INFO: From Linking external/flatbuffers/flatc [for tool]:
ld: warning: ignoring duplicate libraries: '-lm'
ERROR: /Users/tennis/esp/tflite-micro/python/tflite_micro/BUILD:48:17: Linking python/tflite_micro/_runtime.so failed: (Exit 1): cc_wrapper.sh failed: error executing CppLink command (from target //python/tflite_micro:_runtime.so) external/local_config_cc/cc_wrapper.sh @bazel-out/darwin_arm64-fastbuild/bin/python/tflite_micro/_runtime.so-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
ld: Undefined symbols:
  _PyBaseObject_Type, referenced from:
      pybind11::detail::make_object_base_type(_typeobject*) in _runtime.o
  _PyBuffer_Release, referenced from:
      pybind11::buffer_info::~buffer_info() in _runtime.o
  _PyByteArray_AsString, referenced from:
      bool pybind11::detail::string_caster<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, false>::load_raw<char>(std::__1::enable_if<std::is_same<char, char>::value, pybind11::handle>::type) in _runtime.o
  _PyByteArray_Size, referenced from:
      bool pybind11::detail::string_caster<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, false>::load_raw<char>(std::__1::enable_if<std::is_same<char, char>::value, pybind11::handle>::type) in _runtime.o
  _PyByteArray_Type, referenced from:
      bool pybind11::detail::string_caster<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, false>::load_raw<char>(std::__1::enable_if<std::is_same<char, char>::value, pybind11::handle>::type) in _runtime.o
  _PyBytes_AsString, referenced from:
      bool pybind11::detail::string_caster<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, false>::load_raw<char>(std::__1::enable_if<std::is_same<char, char>::value, pybind11::handle>::type) in _runtime.o
  _PyBytes_AsStringAndSize, referenced from:
      pybind11::str::operator std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>() const in _runtime.o
      tflite::ConvertFromPyString(_object*, char**, long*) in python_utils.o
  _PyBytes_FromStringAndSize, referenced from:
      tflite::ConvertToPyString(char const*, unsigned long) in python_utils.o
  _PyBytes_Size, referenced from:
      bool pybind11::detail::string_caster<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, false>::load_raw<char>(std::__1::enable_if<std::is_same<char, char>::value, pybind11::handle>::type) in _runtime.o
  _PyCFunction_Type, referenced from:
      pybind11::cpp_function::initialize_generic(std::__1::unique_ptr<pybind11::detail::function_record, pybind11::cpp_function::InitializingFunctionRecordDeleter>&&, char const*, std::type_info const* const*, unsigned long) in _runtime.o
      PyCFunction_GET_SELF(_object*) in _runtime.o
      PyCFunction_GET_SELF(_object*) in interpreter_wrapper.o
  _PyCMethod_New, referenced from:
      pybind11::cpp_function::initialize_generic(std::__1::unique_ptr<pybind11::detail::function_record, pybind11::cpp_function::InitializingFunctionRecordDeleter>&&, char const*, std::type_info const* const*, unsigned long) in _runtime.o
  _PyCapsule_GetContext, referenced from:
      pybind11::capsule::capsule(void const*, void (*)(void*))::'lambda'(_object*)::operator()(_object*) const in _runtime.o
  _PyCapsule_GetName, referenced from:
      pybind11::capsule::name() const in _runtime.o
      pybind11::capsule::get_name_in_error_scope(_object*) in _runtime.o
  _PyCapsule_GetPointer, referenced from:
      pybind11::detail::internals** pybind11::capsule::get_pointer<pybind11::detail::internals*>() const in _runtime.o
      pybind11::cpp_function::dispatcher(_object*, _object*, _object*) in _runtime.o
      pybind11::detail::function_record* pybind11::capsule::get_pointer<pybind11::detail::function_record>() const in _runtime.o
      pybind11::capsule::capsule(void const*, void (*)(void*))::'lambda'(_object*)::operator()(_object*) const in _runtime.o
      pybind11::detail::type_info* pybind11::capsule::get_pointer<pybind11::detail::type_info>() const in _runtime.o
      _import_array() in numpy_utils.o
  _PyCapsule_New, referenced from:
      pybind11::capsule::capsule(void const*, char const*, void (*)(_object*)) in _runtime.o
      pybind11::capsule::capsule(void const*, void (*)(void*)) in _runtime.o
  _PyCapsule_SetContext, referenced from:
      pybind11::capsule::capsule(void const*, void (*)(void*)) in _runtime.o
[... 348 lines cut ...]
      Py_DECREF(_object*) in _runtime.o
      Py_DECREF(_object*) in interpreter_wrapper.o
      Py_DECREF(_object*) in numpy_utils.o
  __Py_FalseStruct, referenced from:
      pybind11::detail::type_caster<bool, void>::load(pybind11::handle, bool) in _runtime.o
      pybind11::detail::type_caster<bool, void>::cast(bool, pybind11::return_value_policy, pybind11::handle) in _runtime.o
  __Py_NoneStruct, referenced from:
      pybind11::detail::object_api<pybind11::handle>::is_none() const in _runtime.o
      pybind11::none::none() in _runtime.o
      pybind11::detail::object_api<pybind11::detail::accessor<pybind11::detail::accessor_policies::generic_item>>::is_none() const in _runtime.o
  __Py_NotImplementedStruct, referenced from:
      pybind11::cpp_function::dispatcher(_object*, _object*, _object*) in _runtime.o
  __Py_TrueStruct, referenced from:
      pybind11::detail::type_caster<bool, void>::load(pybind11::handle, bool) in _runtime.o
      pybind11::detail::type_caster<bool, void>::cast(bool, pybind11::return_value_policy, pybind11::handle) in _runtime.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Target //python/tflite_micro:whl.dist failed to build
Use --verbose_failures to see the command lines of failed build steps.
ERROR: /Users/tennis/esp/tflite-micro/python/tflite_micro/BUILD:203:9 Copying files to directory failed: (Exit 1): cc_wrapper.sh failed: error executing CppLink command (from target //python/tflite_micro:_runtime.so) external/local_config_cc/cc_wrapper.sh @bazel-out/darwin_arm64-fastbuild/bin/python/tflite_micro/_runtime.so-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
INFO: Elapsed time: 38.234s, Critical Path: 8.51s
INFO: 524 processes: 209 internal, 314 darwin-sandbox, 1 local.
ERROR: Build did NOT complete successfully
```

gamename, 2024-03-03:
mis-understood how this works

closed 2024-03-03 by gamename

khuangCTS, 2024-06-11:
Hi gamename,

I am currently encountering this issue when trying to build on a M1 Mac as well. I was able to sort out my python requirements for Bazel within the third_party folder but now encountering the this CppLink command failure. Do you have any insights or anyway you could point me to the right direction to build the python env? Thanks again

gamename, 2024-06-11:
@khuangCTS 

Sorry guy, its been so long since this happened, I don't remember. :(

khuangCTS, 2024-06-12:
No problem, I was able to figure it out using a different method