tensorflow/tflite-micro #2486: `bazel build //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor` fails

gamename, 2024-03-03:
Hi,

When I run `bazel build`, I get this error:
```
bazel build //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor
Starting local Bazel server and connecting to it...
ERROR: Traceback (most recent call last):
	File "/private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD", line 7, column 16, in <toplevel>
		hdrs = glob(["_site/tensorflow/include/**"], allow_empty=False, exclude_directories=1),
Error in glob: glob pattern '_site/tensorflow/include/**' didn't match anything, but allow_empty is set to False (the default value of allow_empty can be set with --incompatible_disallow_empty_glob).
ERROR: /private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD: no such target '@@tensorflow_cc_deps//:cc_library': target 'cc_library' not declared in package '' defined by /private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD (Tip: use `query "@tensorflow_cc_deps//:*"` to see all the targets in that package)
ERROR: /Users/tennis/esp/tflite-micro/python/tflite_micro/signal/BUILD:255:23: no such target '@@tensorflow_cc_deps//:cc_library': target 'cc_library' not declared in package '' defined by /private/var/tmp/_bazel_tennis/e77d9007a951dc4f16a5de816efa9ff2/external/tensorflow_cc_deps/BUILD (Tip: use `query "@tensorflow_cc_deps//:*"` to see all the targets in that package) and referenced by '//python/tflite_micro/signal:ops/_stacker_op.so'
ERROR: Analysis of target '//tensorflow/lite/micro/examples/micro_speech:audio_preprocessor' failed; build aborted: Analysis failed
INFO: Elapsed time: 5.571s, Critical Path: 0.06s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
FAILED:
    Fetching repository @@tflm_pip_deps_numpy; starting
    Fetching repository @@local_config_xcode; Building xcode-locator
    Fetching repository @@tflm_pip_deps_tensorflow_macos; starting
```

Where does `_site/tensorflow/include/**` come from?  Is there something I need to run first?

Here is my env:
- MacOS 14.2.1 (23C71)
- Chip Apple M1 Max
- Python 3.11.7
- bazel 7.0.2-homebrew

kismeter, 2024-03-03:
@gamename if build on macOS with Apple silicon, you need do some changes
1. change all BUILD file ```requirement("tensorflow")``` to  the ```requirement("tensorflow-macos")```
2. change WORKSPACE file ```http_archive(
    name = "pybind11_bazel",
    sha256 = "044d334a269e03edf4c7f8a3315d1dbf59177b8ddf1c38178bcc72cc73a93aba",
    strip_prefix = "pybind11_bazel-2.11.1.bzl.1",
    urls = ["https://github.com/pybind/pybind11_bazel/releases/download/v2.11.1.bzl.1/pybind11_bazel-2.11.1.bzl.1.zip"],
)``` and ```http_archive(
    name = "pybind11",
    build_file = "@pybind11_bazel//:pybind11.BUILD",
    sha256 = "d475978da0cdc2d43b73f30910786759d593a9d8ee05b1b6846d1eb16c6d2e0c",
    strip_prefix = "pybind11-2.11.1",
    urls = ["https://github.com/pybind/pybind11/archive/refs/tags/v2.11.1.tar.gz"],
)```
3. you might need change [signal/tensorflow_core/kernels/overlap_add_kernel.cc#](https://github.com/tensorflow/tflite-micro/blob/main/signal/tensorflow_core/kernels/overlap_add_kernel.cc#L46) to ```memset(state_tensors_[i].template flat<T>().data(), 0, sizeof(T) * frame_size_);``` and https://github.com/tensorflow/tflite-micro/blob/main/signal/tensorflow_core/kernels/overlap_add_kernel.cc#L59 to ```T* buffer = state_tensors_[i].template flat<T>().data();```

gamename, 2024-03-03:
Hi @kismeter

Thanks for the response. Getting the same error after the suggested changes.
Again, what creates the `_site/tensorflow/include` directories?

```bazel build //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor
ERROR: Traceback (most recent call last):
	File "/private/var/tmp/_bazel_tennis/99f6b3c6b91c4dd4a19bef201b9505ad/external/tensorflow_cc_deps/BUILD", line 7, column 16, in <toplevel>
		hdrs = glob(["_site/tensorflow/include/**"], allow_empty=False, exclude_directories=1),
Error in glob: glob pattern '_site/tensorflow/include/**' didn't match anything, but allow_empty is set to False (the default value of allow_empty can be set with --incompatible_disallow_empty_glob).
ERROR: /private/var/tmp/_bazel_tennis/99f6b3c6b91c4dd4a19bef201b9505ad/external/tensorflow_cc_deps/BUILD: no such target '@@tensorflow_cc_deps//:cc_library': target 'cc_library' not declared in package '' defined by /private/var/tmp/_bazel_tennis/99f6b3c6b91c4dd4a19bef201b9505ad/external/tensorflow_cc_deps/BUILD (Tip: use `query "@tensorflow_cc_deps//:*"` to see all the targets in that package)
INFO: Repository pybind11 instantiated at:
  /Users/tennis/src/tflite-micro-fork/WORKSPACE:64:13: in <toplevel>
Repository rule http_archive defined at:
  /private/var/tmp/_bazel_tennis/99f6b3c6b91c4dd4a19bef201b9505ad/external/bazel_tools/tools/build_defs/repo/http.bzl:381:31: in <toplevel>
ERROR: /Users/tennis/src/tflite-micro-fork/python/tflite_micro/signal/BUILD:134:23: no such target '@@tensorflow_cc_deps//:cc_library': target 'cc_library' not declared in package '' defined by /private/var/tmp/_bazel_tennis/99f6b3c6b91c4dd4a19bef201b9505ad/external/tensorflow_cc_deps/BUILD (Tip: use `query "@tensorflow_cc_deps//:*"` to see all the targets in that package) and referenced by '//python/tflite_micro/signal:ops/_filter_bank_ops.so'
ERROR: Analysis of target '//tensorflow/lite/micro/examples/micro_speech:audio_preprocessor' failed; build aborted: Analysis failed
INFO: Elapsed time: 2.339s, Critical Path: 0.05s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
FAILED:
    Fetching repository @@tflm_pip_deps_tensorflow_macos; starting
    Fetching repository @@tflm_pip_deps_numpy; starting
    Fetching https://github.com/pybind/pybind11/archive/refs/tags/v2.11.1.tar.gz; 596.4 KiB (80.7%)```

kismeter, 2024-03-03:
also change tensorflow-macos==2.15.0 in the file third_party/python_requirements.in and run bazel run //third_party:python_requirements.update

kismeter, 2024-03-03:
```bazel run //third_party:python_requirements.update -- --upgrade```  might also needed

gamename, 2024-03-03:
> 3\. T* buffer = state_tensors_[i].template flat<T>().data();

Thanks! I'll give it a try.

gamename, 2024-03-03:
Hi @kismeter 

It worked! Thank you!

```bazel build //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor
Starting local Bazel server and connecting to it...
INFO: Analyzed target //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor (124 packages loaded, 22309 targets configured).
INFO: From Linking external/ruy/ruy/profiler/libinstrumentation.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: archive library: bazel-out/darwin_arm64-fastbuild/bin/external/ruy/ruy/profiler/libinstrumentation.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking tensorflow/lite/kernels/internal/libtensor_utils_no_eigen.a:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: warning renaming duplicate member name 'portable_tensor_utils.o' from 'bazel-out/darwin_arm64-fastbuild/bin/tensorflow/lite/kernels/internal/_objs/tensor_utils_no_eigen/0/portable_tensor_utils.o(portable_tensor_utils.o)' and 'bazel-out/darwin_arm64-fastbuild/bin/tensorflow/lite/kernels/internal/_objs/tensor_utils_no_eigen/1/portable_tensor_utils.o(portable_tensor_utils.o)'
INFO: From Linking external/flatbuffers/flatc [for tool]:
ld: warning: ignoring duplicate libraries: '-lm'
INFO: Found 1 target...
Target //tensorflow/lite/micro/examples/micro_speech:audio_preprocessor up-to-date:
  bazel-bin/tensorflow/lite/micro/examples/micro_speech/audio_preprocessor
INFO: Elapsed time: 45.609s, Critical Path: 8.38s
INFO: 597 processes: 232 internal, 365 darwin-sandbox.
INFO: Build completed successfully, 597 total actions```

closed 2024-03-03 by gamename