tensorflow/tflite-micro #2499: Cannot fetch required TF version 2.15.0.post1 for building  TensorFlow Lite Micro example hello_world:

tjpeter, 2024-03-12:
When attempting to build the TensorFlow Lite Micro example hello_world:evaluate given in
[https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/README.md#](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/README.md#), the build process fails due to an error fetching the required TensorFlow version. The error message (see below) indicates that the specified version 2.15.0.post1 cannot be found in the available distributions.

**Platform**:
macOS: 13.4.1
Python Platform: macOS-13.4.1-arm64-arm-64bit
Python 3.10.13 (main, Sep 11 2023, 08:16:02) [Clang 14.0.6 ]

**Steps to Reproduce**:
Execute the command: 'bazel build tensorflow/lite/micro/examples/hello_world:evaluate'

**Expected Behavior**:
The build process completes successfully, fetching the required TensorFlow version 2.15.0.post1.

```
bazel build tensorflow/lite/micro/examples/hello_world:evaluate

INFO: Repository tflm_pip_deps_tensorflow instantiated at:
  /Users/johndoe/Documents/projects/tflite-micro/WORKSPACE:55:13: 
  in <toplevel>
  /private/var/tmp/_bazel_johndoe/298a0a50674461cbee601d6aa9c776b2/external/tflm_pip_deps/requirements.bzl:49:20: 
  in install_deps
   Repository rule whl_library defined at:
  /private/var/tmp/_bazel_johndoe/298a0a50674461cbee601d6aa9c776b2/external/rules_python/python/pip_install/pip_repository.bzl:715:30: in <toplevel>
ERROR: An error occurred during the fetch of repository 'tflm_pip_deps_tensorflow':
   Traceback (most recent call last):
    File "/private/var/tmp/_bazel_johndoe/298a0a50674461cbee601d6aa9c776b2/external/rules_python/python/pip_install/pip_repository.bzl", line 691, column 13, in _whl_library_impl
                fail("whl_library %s failed: %s (%s) error code: '%s'" % (rctx.attr.name, result.stdout, result.stderr, result.return_code))
Error in fail: whl_library tflm_pip_deps_tensorflow failed:  
(ERROR: Could not find a version that satisfies the requirement tensorflow==2.15.0.post1 (from versions: 2.13.0rc0, 2.13.0rc1, 2.13.0rc2, 2.13.0, 2.13.1, 2.14.0rc0, 2.14.0rc1, 2.14.0, 2.14.1, 2.15.0rc0, 2.15.0rc1, 2.15.0, 2.15.1, 2.16.0rc0, 2.16.1)
ERROR: No matching distribution found for tensorflow==2.15.0.post1
```

whaleygeek, 2024-03-14:
I have the same issue here on my Mac. It looks like we are falling between an emerging dependency nightmare that has just surfaced recently. All the Shawn Hymell arduino examples on his blog also seem to be broken. 

I followed the install instructions for bazel and it upgraded my python to 3.12, and none of the pip dependency resolution in the tflite hello world worked, failing due to deleted setuptools methods between 3.11 and 3.12.

I manually moved my brew link to 3.11.8 and it solved that issue, but now I have exactly the same issue as above, where the hello world bazel build will not find a matching distribution for tensorflow.

whaleygeek, 2024-03-18:
If I change between Python 3.12, 3.11 and 3.10 I get different behaviour. Python 3.12 complains about a missing method inside setuptools, Python 3.11 and earlier complain about no suitable candidate.

On Ubuntu I get a different issue, the wheel for numpy is unavailable and it tries to compile from C, and it complains about <Python.h> missing.

mansnils, 2024-03-20:
Hi,
I think the reason may be that the 2.15.0.post1 package is only released for Linux x86_64, according to a recent TF blog post..
https://blog.tensorflow.org/2023/12/tensorflow-215-update-hot-fix-linux-installation-issue.html

Does it help running: `bazel run //third_party:python_requirements.update -- --upgrade` ?

ctwillson, 2024-03-20:
same issue

whaleygeek, 2024-03-22:
Well, that gets over that specific problem (thanks for that!), and it spends some time downloading dependencies and building lots of cpp code. But then the build fails (small extract below...) - I suspect this might be some python runtime version issue, as a new version of Python has recently broken lots of other projects of mine, and I am repeatedly switching between different python distributions. I will try a few different versions of Python and see if any other version works.

  "__Py_TrueStruct", referenced from:
      pybind11::detail::type_caster<bool, void>::load(pybind11::handle, bool) in _runtime.o
      pybind11::detail::type_caster<bool, void>::cast(bool, pybind11::return_value_policy, pybind11::handle) in _runtime.o
      pybind11::detail::type_caster<bool, void>::load(pybind11::handle, bool) in interpreter_wrapper.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Target //tensorflow/lite/micro/examples/hello_world:evaluate failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 1.256s, Critical Path: 0.65s
INFO: 2 processes: 2 internal.
ERROR: Build did NOT complete successfully

tjpeter, 2024-04-02:
I have tried the suggestion from @mansnils 

However, I still fails, albeit with different issues, referring to the arm64 architecture. Small extract below:

```
INFO: From Linking tensorflow/lite/kernels/internal/libtensor_utils_no_eigen.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: warning duplicate member name 'portable_tensor_utils.o' from 'bazel-out/darwin_arm64-fastbuild/bin/tensorflow/lite/kernels/internal/_objs/tensor_utils_no_eigen/0/portable_tensor_utils.o(portable_tensor_utils.o)' and 'bazel-out/darwin_arm64-fastbuild/bin/tensorflow/lite/kernels/internal/_objs/tensor_utils_no_eigen/1/portable_tensor_utils.o(portable_tensor_utils.o)'
INFO: From Linking external/flatbuffers/flatc [for tool]:
ld: warning: ignoring duplicate libraries: '-lm'
ERROR: /Users/JohnDoe/Documents/projects/tflite-micro/python/tflite_micro/BUILD:48:17: Linking python/tflite_micro/_runtime.so failed: (Exit 1): cc_wrapper.sh failed: error executing CppLink command (from target //python/tflite_micro:_runtime.so) external/local_config_cc/cc_wrapper.sh @bazel-out/darwin_arm64-fastbuild/bin/python/tflite_micro/_runtime.so-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
Undefined symbols for architecture arm64:
  "_PyBaseObject_Type", referenced from:
      pybind11::detail::make_object_base_type(_typeobject*) in _runtime.o
  "_PyBuffer_Release", referenced from:
      pybind11::buffer_info::~buffer_info() in _runtime.o
```

...

mansnils, 2024-04-02:
I have not seen the issues reported in the last two comments. Don't know if it matters but I am using python 3.10.

closed 2024-05-03 by github-actions[bot]

enytz, 2024-06-17:
I've been trying to put together a project for over 3 days now... There is a file in a separate Tensorflow micro repository https://github.com/tensorflow/tflite-micro/blob/main/third_party/python_requirements.txt where version 2.15.0.post1 is specified, but it does not exist either in the main repository or in micro, in order to remove this problem, I installed pip-compile and used it to rebuild this file, the correct hash and version 2.15.0 were added, the project began to assemble further, but already went other mistakes... But that's another story.