tensorflow/tflite-micro #2518: Usage of NaN and Infinity not supported by default in armclang

AdrianLundell, 2024-03-25:
Using these is not supported by the std/fast ffp-mode in armclang, forcing useage of the --ffp-mode=full flag.
This is fixed for all armclang builds with #2517, but it could be worth the effort to look into supporting other ffp-modes, in particular since they may be faster.

See https://developer.arm.com/documentation/101754/0622/armclang-Reference/armclang-Command-line-Options/-ffp-mode

rascani, 2024-04-02:
@AdrianLundell - is this fixed now? If so, please close.

AdrianLundell, 2024-04-03:
It is fixed to the extent that we do not build anything with undefined behaviour anymore since #2517, but the issue has a broader scope to add support for all fpp-modes in the future, so it would be nice to keep around for that if that is ok.

closed 2024-06-22 by github-actions[bot]