tensorflow/tflite-micro #2523: ARMCM4.h: No such file or directory when building for Target = cortex_m_generic and Target_Arch=cortex-m4

Urmi-Jana, 2024-03-27:
While attempting to build the TFLM library for the Cortex-M4 platform without CMSIS optimizations with the instructions given in [https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md](), the build process fails due to the ARMCM4.h file not existing. The error message (see below) indicates that the ARMCM4.h file is required for micro_time.o file

**Platform:**
Ubuntu 20.04
Python: 3.8
Tensorflow: 2.12.0 

**Steps to reproduce:**
Execute the command `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m4 microlite`

**Expected behavior:**
The build process executes successfully, producing the appropriate library file to be included in my project

```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m4 microlite

tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/gcc_embedded already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/arm-none-eabi-g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Wnon-virtual-dtor -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wunused-variable -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wmissing-field-initializers -Wstrict-aliasing -Wno-unused-parameter  -DKERNELS_OPTIMIZED_FOR_SIZE -mcpu=cortex-m4+nofp -mfpu=fpv4-sp-d16 -DTF_LITE_MCU_DEBUG_LOG -mthumb -mfloat-abi=soft -funsigned-char -mlittle-endian -Wno-type-limits -Wno-unused-private-field -fomit-frame-pointer -MD -DCPU_M4=1 -DCMSIS_DEVICE_ARM_CORTEX_M_XX_HEADER_FILE=\""ARMCM4".h\" -D"ARMCM4" -O0 -I. -Itensorflow/lite/micro/tools/make/downloads -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/kissfft -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/downloads/cmsis/Device/ARM/"ARMCM4"/Include -Itensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/Core/Include -Igen/cortex_m_generic_cortex-m4_default_gcc/genfiles/ -Igen/cortex_m_generic_cortex-m4_default_gcc/genfiles/ -c tensorflow/lite/micro/cortex_m_generic/micro_time.cc -o gen/cortex_m_generic_cortex-m4_default_gcc/obj/core/tensorflow/lite/micro/cortex_m_generic/micro_time.o
<command-line>: fatal error: ARMCM4.h: No such file or directory
compilation terminated.
make: *** [tensorflow/lite/micro/tools/make/Makefile:795: gen/cortex_m_generic_cortex-m4_default_gcc/obj/core/tensorflow/lite/micro/cortex_m_generic/micro_time.o] Error 1
```

Any help would be highly appreciated!!

mansnils, 2024-03-28:
Closing as it should be working now.

closed 2024-03-28 by mansnils

Urmi-Jana, 2024-04-11:
Hi @mansnils , thanks so much for the fix, the make command works perfectly now.

I wanted to clarify that this makes the CMSIS files by default, even it is not specified on the make command. Is it not possible to build the TFLM library completely without CMSIS?

mansnils, 2024-04-11:
That's a good point, as it should be possible to build the TFLM library without CMSIS for this target. Reason being is that a CMSIS header is needed for performance counters: https://github.com/tensorflow/tflite-micro/blob/7aa0d93dcef1bba137c742836aed444cf7c853ef/tensorflow/lite/micro/cortex_m_generic/micro_time.cc#L19
The alternative would be to remove the support for those counters.

Urmi-Jana, 2024-04-12:
So technically, CMSIS is absolutely required to use TFLM on cortex-m4, since, again, those counters are needed for multiple use-cases