tensorflow/tflite-micro #2524: The difference between tensorflow lite and tensorflow lite micro

ctwillson, 2024-03-27:
I read the doc about the operators difference between TensorFlow Lite and  TensorFlow 
[image]
However,but I have no idea about the difference between TensorFlow Lite and  TensorFlow Lite micro

rascani, 2024-03-27:
The difference between TensorFlow Lite and TensorFlow Lite for Microcontrollers is the platforms they target. TFLite is targeted at mobile use cases, such as Android & iOS devices. As the name suggests, TFLite-Micro is aimed at low-power microcontrollers and digital signal processors for embedded use cases. These can even be baremetal use cases, as TFLM has no dependencies on an operating system.

closed 2024-03-27 by rascani

ctwillson, 2024-03-28:
> The difference between TensorFlow Lite and TensorFlow Lite for Microcontrollers is the platforms they target. TFLite is targeted at mobile use cases, such as Android & iOS devices. As the name [...]

Are the operators supported by TF lite same as TF Microcontrollers?In other words,if I have .tflite,I used cmd like "xxd -i converted_model.tflite > model_data.cc" and the microcontrollers have enough memories,so it must can be loaded normally?

rascani, 2024-03-28:
TFLM supports the same .tflite model format as TFLite. However, there are some operators that are unimplemented. For the most part, these can be ported from TFLite as needed and there is a [guide](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md) on how to do so.