tensorflow/tflite-micro #2526: ERROR: An error occurred during the fetch of repository 'tflm_pip_deps_tensorflow':

ctwillson, 2024-03-28:
bazel build tensorflow/lite/micro/examples/hello_world:evaluate_test failed:
```
INFO: Repository tflm_pip_deps_tensorflow instantiated at:
  /home/bsp/tflite-micro/WORKSPACE:55:13: in <toplevel>
  /home/bsp/.cache/bazel/_bazel_bsp/3af90413583204447fe7eb7c72beb4a7/external/tflm_pip_deps/requirements.bzl:49:20: in install_deps
Repository rule whl_library defined at:
  /home/bsp/.cache/bazel/_bazel_bsp/3af90413583204447fe7eb7c72beb4a7/external/rules_python/python/pip_install/pip_repository.bzl:715:30: in <toplevel>
ERROR: An error occurred during the fetch of repository 'tflm_pip_deps_tensorflow':
   Traceback (most recent call last):
	File "/home/bsp/.cache/bazel/_bazel_bsp/3af90413583204447fe7eb7c72beb4a7/external/rules_python/python/pip_install/pip_repository.bzl", line 691, column 13, in _whl_library_impl
		fail("whl_library %s failed: %s (%s) error code: '%s'" % (rctx.attr.name, result.stdout, result.stderr, result.return_code))
Error in fail: whl_library tflm_pip_deps_tensorflow failed:  (Timed out) error code: '256'
ERROR: /home/bsp/tflite-micro/WORKSPACE:55:13: fetching whl_library rule //external:tflm_pip_deps_tensorflow: Traceback (most recent call last):
	File "/home/bsp/.cache/bazel/_bazel_bsp/3af90413583204447fe7eb7c72beb4a7/external/rules_python/python/pip_install/pip_repository.bzl", line 691, column 13, in _whl_library_impl
		fail("whl_library %s failed: %s (%s) error code: '%s'" % (rctx.attr.name, result.stdout, result.stderr, result.return_code))
Error in fail: whl_library tflm_pip_deps_tensorflow failed:  (Timed out) error code: '256'
ERROR: no such package '@@tflm_pip_deps_tensorflow//': whl_library tflm_pip_deps_tensorflow failed:  (Timed out) error code: '256'
ERROR: /home/bsp/tflite-micro/tensorflow/lite/micro/examples/hello_world/BUILD:46:10: //tensorflow/lite/micro/examples/hello_world:evaluate depends on @@tflm_pip_deps_tensorflow//:pkg in repository @@tflm_pip_deps_tensorflow which failed to fetch. no such package '@@tflm_pip_deps_tensorflow//': whl_library tflm_pip_deps_tensorflow failed:  (Timed out) error code: '256'
ERROR: Analysis of target '//tensorflow/lite/micro/examples/hello_world:evaluate_test' failed; build aborted: Analysis failed
INFO: Elapsed time: 643.348s, Critical Path: 0.02s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
```
My ENV:
- ubuntu:22.04
- Python 3.10.12
- TF:2.16
- bazel:7.1.1

closed 2024-04-28 by github-actions[bot]

haozh960, 2024-10-11:
I have met the same error,  is there any suggestion ?

ctwillson, 2024-10-11:
> @ctwillson I have met the same error, is there any suggestion ?

I am sorry I fogot how I fixed it, so pls  check your code...

ilya-kupchenko, 2025-08-04:
To whoever runs into such errors as timeouts while downloading 'tflm_pip_deps_tensorflow'. These two packages: 'tflm_pip_deps_tensorflow' and 'tensorflow_cc_deps' are the biggest ones and can weight hundreds of megabytes (it is almost the full Tensorflow package AFAIK). So it is timing out simply because it takes too long to download them, i.e. you connection speed is too slow for the timeout cutoff value. Try a larger timeout scaling:

`bazel build --http_timeout_scaling=15.0 --experimental_scale_timeouts=15.0 [your_taget]`

This worked for me.