tensorflow/tflite-micro #2529: Can't get fused as a SVDF operator when convert to TFLM

kismeter, 2024-03-31:
There're some models include SVDF implemented, for example [create_low_latency_svdf_mode](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/examples/speech_commands/models.py#L458) in [tensorflow/speech_commands](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/examples/speech_commands). and also [svdf_conv](https://github.com/google/model_search/blob/master/model_search/ops/svdf_conv.py). when using those implemented SVDF layer to build a model then convert to TFLM, the SVDF won't get fused as a SVDF operator, only got {'CONV_2D', 'SUM', 'FULLY_CONNECTED', 'RESHAPE', 'PAD', 'SOFTMAX', 'DEPTHWISE_CONV_2D', 'ADD'}, is there any changes should be done to get fused to SVDF operator? or the SVDF operator is won't support in the future?

rascani, 2024-04-02:
There are no plans to remove the SVDF operator. I'm not sure why those examples are no longer getting fused into SVDF, but its possible that other converter passes are preventing that. I'll leave this open to look into further later.

ddavis-2015, 2024-04-26:
@kismeter 

This problem has been traced to an issue with `TFLiteConverter`, and is tracked in the TensorFlow repository as [issue #63035](https://github.com/tensorflow/tensorflow/issues/63035).

This issue will now be closed.  Please feel free to reopen the issue should you have additional comments.

closed 2024-04-26 by ddavis-2015