tensorflow/tflite-micro #2535: Official Release of tflite-micro

raveendrapadala, 2024-04-02:
Team,

- Glad to see the community is contributing actively. Are there plans to release official release for this project?
- Are there plans to integrate this project to Yocto meta-machinelearning layer? i see tensorflow-lite (https://layers.openembedded.org/layerindex/recipe/105004/) so was the request
- https://layers.openembedded.org/layerindex/branch/master/layer/meta-machinelearning/

rascani, 2024-04-02:
No, we don't have any plans to introduce an official release for this project. Nor do we have any plans to add its use in Yocto, though others are welcome to do so.

closed 2024-04-02 by rascani

raveendrapadala, 2024-04-29:
Understood. Thanks. Will share the bitbake files.