tensorflow/tflite-micro #2537: flatbuffers/flatbuffers.h

SethARhodes, 2024-04-03:
I am trying to build a simple LSTM project in Zephyr and I got a bunch of errors where the tflite-micro source files can't find flatbuffers/flatbuffers.h.

I tried building tensorflow lite from source with CMake per the instructions here: https://www.tensorflow.org/lite/guide/build_cmake

That produced the flatbuffers folder containing the flatbuffers.h file but the other problem is that if I copy that folder over to my project/modules/lib/tflite-micro/tensorflow/lite/micro folder, it finds the header, but then all the files in that flatbuffers folder try to include "flatbuffers/{something}" but that fails since it can't find yet another flatbuffers folder. I tried going through and deleting the "flatbuffers/" portion from all the includes and it still fails to build. 

I'm pulling my hair out here just trying to deploy my tflm project.

rascani, 2024-04-03:
TFLite and TFLM do share some code, but they are different projects. Notably, TFLM does not have a CMake build. We support Bazel & GNU Make. Both of those build systems are configured to download Flatbuffers into a location the rest of the build system expects. I would also point out that we add a [patch](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/make/flatbuffers.patch) to flatbuffers to eliminate possible dynamic memory allocation.

You may also want to look at the [create_tflm_tree.py](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py) script. This script can be used to generate a "fresh" tflm source tree for a given configuration.

SethARhodes, 2024-04-03:
Can you tell me exactly how to use Bazel to do this? I'm not familiar with the tool and I just need to get this model running on a microcontroller.