tensorflow/tflite-micro #2574: Flashing code to sparkfun edge

h-tye, 2024-05-16:
Hello, I am trying to upload example code to the SparkFun Edge and have been following these instructions: https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/#4 
I ran into issues at step 4 when trying to execute:
```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge micro_speech_bin
```
After a lot of digging, I found this on issue #341:

The following command will build all the examples:
```
make examples
```

_Originally posted by @advaitjain in https://github.com/tensorflow/tflite-micro/issues/341#issuecomment-902321663_

and it did not throw any errors. However, I am unsure what it actually did. Has it completely built all the necessary binary so that I will be able to flash it to the board? Any help would be much appreciated.

ddavis-2015, 2024-05-20:
@h-tye 

If you are using the https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples repository, then the `make examples` command builds all the example applications.  The binaries for those applications are located in the `gen/bin/` directory relative to your cloned repository root.

As for uploading a binary to your SparkFun board, you can try the instructions starting at step 5 (https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/#4) or refer to any instructions given for your specific board at the SparkFun website.

Please note there is no official TFLM support for this platform.

closed 2024-05-20 by ddavis-2015

h-tye, 2024-05-20:
Thanks so much. Just one more question, in the gen/bin/ directory, the binaries are labeled "hello_world.d" and "micro_speech.d" rather than .bin. In the instructions(https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/#4), .bin files are used, do you know if .d files can be used interchangeably with .bin? Or is this just something I'll have to experiment with?