tensorflow/tflite-micro #2576: GreedyMemoryPlanner::operator delete(void *) is private

longhoangth, 2024-05-17:
src/tensorflow/lite/micro/micro_allocator.cpp:390:55: error: 'static void tflite::GreedyMemoryPlanner::operator delete(void*)' is private within this context

longhoangth, 2024-05-17:
build with [tflite-micro-arduino-examples](https://github.com/tensorflow/tflite-micro-arduino-examples)

ddavis-2015, 2024-05-18:
@longhoangth 

Can you provide a much more detailed description of your issue?

longhoangth, 2024-05-18:
When I modify the arduino tensorflow lite library to work with esp32s3 on platformio, the build log error is in image below.
[image]

rascani, 2024-05-20:
This is by design. The `GreedyMemoryPlanner`, like several other TFLM objects, is designed to be allocated from the arena provided to the MicroInterpreter. It is typically allocated here: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_allocator.cc#L88-L90. Notice how it is constructed with placement new.

closed 2024-05-23 by longhoangth

tanmanh0707, 2025-08-30:
> When I modify the arduino tensorflow lite library to work with esp32s3 on platformio, the build log error is in image below. [image]

Hello, I run into the same issue. How did you resolve this? Thank you.