tensorflow/tflite-micro #2593: Passing custom/additional data to kernels

DanieleParravicini-Synthara, 2024-05-22:
Hello, I have the following use case that I would like to cover.

How can I provide an efficient kernel for a layer without breaking the compatibility of the model?

Consider:

1. a layer of a NN model (e.g. a linear layer) that could be accelerated by a dedicated hardware module. 
E.g. imagine you have M cores and you break the linear layers in M chunks.

2.  the custom hardware module could benefit from additional information (e.g. how many elements each core could use).

Here follows some ideas:

1. use custom operator

    1. for each layer that you want to accelerate create a custom operator
    2. modify the NN model replacing the original layer with a custom operator that is functionally equivalent to the first one
    3. retrain the NN model
    4. convert the model to a TensorFlow Lite Model adding the additional custom operator (and possibly passing additional info to the new custom layer using `custom_options` field of Operator of flatbuffer)
    5. provide a implementation of the operator to the interpreter that reads the `custom_options` field and execute the layer accordingly

4. keep the same operator and use a custom conversion tools that adds information on the `custom_options` field. In this way I can avoid the first three steps of the previous list and do the following
    1.    convert the model to a TensorFlow Lite Model adding the additional custom operator and passing additional info to the new custom layer using `custom_options` field of Operator of flatbuffer.
       e.g. like this
```
import np
import tensorflow as tf
from tensorflow.lite.python import schema_py_generated as schema_fb

def load_model(save_path: str):
    with open(save_path, "rb") as f:
       return f.read()

tflite_quantized = load_model("models/mlp_int8.tflite")

aModel = schema_fb.ModelT.InitFromPackedBuf(tflite_quantized, 0)
     

def BuiltinCodeToName(code):
  """Converts a builtin op code enum to a readable name."""
  for name, value in schema_fb.BuiltinOperator.__dict__.items():
    if value == code:
      return name
  return None
     

for i, op in enumerate(aModel.subgraphs[0].operators):
    op_code = aModel.operatorCodes[op.opcodeIndex].builtinCode
    print(f"[{i}] :  {BuiltinCodeToName(op_code)} ({op_code})")

     
### FROM HERE
custo = np.ones(10,dtype=np.uint8)
aModel.subgraphs[0].operators[4].customOptions = custo
### TO HERE

from tflite_support import flatbuffers
b = flatbuffers.Builder(0)

b.Finish(
    aModel.Pack(b)
)
model_buff = b.Output()

def save_tflite_model(tflite_model, save_dir, model_name):
	"""save the converted tflite model
	Args:
		tflite_model (binary): the converted model in serialized format.
		save_dir (str): the save directory
		model_name (str): model name to be saved
	"""
    import os

	if not os.path.exists(save_dir):
		os.makedirs(save_dir)
	save_path = os.path.join(save_dir, model_name)
	
	with open(save_path, "wb") as f:
		f.write(tflite_model)

save_tflite_model(model_buff, "MLP_models", "mlp_int8.tflite" )
```

   2.  I will have to modify the behaviour of [microinterpreter](https://github.com/tensorflow/tflite-micro/blob/dfdd666ae075e07da288ce8d4a38c60aeecbc07a/tensorflow/lite/micro/micro_interpreter.cc#L157C9-L157C47) to allow for builtin operators to have custom_options
   3.  provide a suitable implementation of the operator that uses the custom_options of the operator to do something smart
 

with the second approach I see the following advantages:
- I do not have to modify the model. I can just write a python script that could look at the model and "annotate the layers with custom_options" when needed.
- I have compatibility with the original model and can switch between accelerated and non accelerated kernels (e.g. in certain cases due to the fixed costs needed to start the dedicated hardware module the reference implementation or another operator implementation is better suited) 
- I lower the complexity of accelerating operators
- No need to retrain the model from scratch

DanieleParravicini-Synthara, 2024-05-22:
it might be related to https://github.com/tensorflow/tflite-micro/issues/619

DanieleParravicini-Synthara, 2024-05-24:
Thank you for your reply

Makes a lot of sense to do a AOT that could emit custom operators. I suspected that the option to go was either that or to modify the model with CustomOPs.

You are right, the Metadata could be a way and I tried to use it, but how could we map the specific Operator to a key that could then be used to access the Metadata?
Is there an Id of the operator that one could use to access the metadata key/value store? Or something similar?
The `custom_options`  seem already there and available to use.

DanieleParravicini-Synthara, 2024-05-28:
Wouldn't it be as simple as modifying this [line](https://github.com/tensorflow/tflite-micro/blob/dfdd666ae075e07da288ce8d4a38c60aeecbc07a/tensorflow/lite/micro/micro_interpreter.cc#L157C9-L157C47)  to give all the operators access to `custom_options` so as to enable information exchange from an AOT tool?

rascani, 2024-05-31:
My reluctance to using `custom_options` is because it is intended for a different purpose than this. That field is supposed to be the parameters for a custom op, not a built-in. Allowing `custom_options` to be populated when using a builtin operator is breaking that abstraction. It wouldn't be a change I could take upstream.

You're looking for an extension point, which is what `Metadata` was designed for. This would actually also require some additional changes to TFLM. We've used Metadata before for exactly this use case in TFLite, but hadn't run into it with TFLM before. We'd probably want to extend the `MicroContext` or `MicroGraph` for providing access to subgraph_id and node_id, which we could then use to lookup op specific parameters within a Metadata buffer.

DanieleParravicini-Synthara, 2024-06-03:
Thanks @rascani.
I think that now I understand your point `custom_options` is only for builtin.

I think we could help propagating the subgraph_id and node_id to the evaluation context in the node. 
Can we help?

rascani, 2024-06-03:
That would be great!

Taking a second look, it looks like subgraph index is already available from the [`MicroGraph::GetCurrentSubgraphIndex()`](https://github.com/tensorflow/tflite-micro/blob/71fba59b5bebfd64b48e167bfcb669fff568ec74/tensorflow/lite/micro/micro_interpreter_graph.h#L87) function. Kernels have access to the Graph via `tflite::GetMicroContext(context)->graph()`.

I think I would lean towards a similar API for node id, but the thing to watch for is that kernels can invoke subgraphs. We'll probably need a similar scheme as `current_subgraph_idx_` & `previous_subgraph_idx`.

DanieleParravicini-Synthara, 2024-06-04:
If I can ask, why subgraph_idx is size_t whereas current_subgraph_index_ and previous_subgraph_idx are int?

What  is the objective?

DanieleParravicini-Synthara, 2024-06-04:
To start, something like https://github.com/tensorflow/tflite-micro/pull/2605 ?

closed 2024-07-02 by rascani