tensorflow/tflite-micro #2619: Feature request: Update Ethos-U driver ?

iabdalkader, 2024-07-07:
Hi! It seems that the Ethos-U core driver currently used by tflite-micro is somewhat outdated. It's missing the `ethosu_address_remap` callback, which was introduced around release v23.02 (the current release is 24.05). This function is crucial; without it, invocation will fail if the NPU core cannot use addresses directly. 

Additionally, could you please consider allowing configuration arguments (such as the ones defined in `src/ethosu_config_uxx.h`) to be passed to the driver from the command line when building the library?

Thanks!

rascani, 2024-07-08:
@mansnils

mansnils, 2024-07-11:
Hi @iabdalkader 
Thanks for your interest in this. We shall update this example. Also keep in mind that contributions are welcome.

iabdalkader, 2024-07-16:
https://github.com/tensorflow/tflite-micro/pull/2626

PR #2626 (2024-07-24): Update Ethos-U driver.
* Update Ethos-U driver to the latest (v24.05)
* Allow passing extra C flags to the driver from the command line.
* Fixes #2619

Note I tested with `TARGET=cortex_m_corstone_300 test_network_tester_test` and on real hardware as well built with:
```bash
make -j12 -f tensorflow/lite/micro/tools/make/Makefile \
TARGET=cortex_m_generic TARGET_ARCH=cortex-m55 \
CO_PROCESSOR=ethos_u ETHOSU_ARCH=u55 OPTIMIZED_KERNEL_DIR=ethos_u  \
CORE_OPTIMIZATION_LEVEL=-O2 KERNEL_OPTIMIZATION_LEVEL=-O2 \
THIRD_PARTY_KERNEL_OPTIMIZATION_LEVEL=-O2 \
TARGET_TOOLCHAIN_ROOT=/opt/arm-none-eabi/bin/ [... cut]
commit 213eb876 (2024-07-24): Update Ethos-U driver. (#2626)

* Update Ethos-U driver to the latest (v24.05)
* Allow passing extra C flags to the driver from the command line.
* Fixes #2619

Note I tested with `TARGET=cortex_m_corstone_300 test_network_tester_test` and on real hardware as well built with:
```bash
make -j12 -f tensorflow/lite/micro/tools/make/Makefile \
TARGET=cortex_m_generic TARGET_ARCH=cortex-m55 \
CO_PROCESSOR=ethos_u ETHOSU_ARCH=u55 OPTIMIZED_KERNEL_DIR=ethos_u  \
CORE_OPTIMIZATION_LEVEL=-O2 KERNEL_OPTIMIZATION_LEVEL=-O2 \
THIRD_PARTY_KERNEL_OPTIMIZATION_LEVEL=-O2 \
TARGET_TOOLCHAIN_ROOT=/opt/arm-none- [... cut]
closed 2024-07-24 by mergify[bot]