tensorflow/tflite-micro #2622: TFLITE on ARM64

devapriyas2001, 2024-07-10:
Hi,
I tried to build TFlite for ARM64 using the instructions provided in https://www.tensorflow.org/lite/guide/build_cmake_arm#build_for_aarch64_arm64 . I was able to successfully configure, but while building using `make-j4` , build errors are produced. Please provide instructions on how to build.

devapriyas2001, 2024-07-10:
Line  886: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:39:9: error: unknown type name ‘bfloat16_t’
	Line  889: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:39:33: error: unknown type name ‘bfloat16_t’
	Line  892: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:40:3: error: unknown type name ‘bfloat16_t’; did you mean ‘float16_t’?
	Line  898: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:40:21: error: ‘bfloat16_t’ undeclared (first use in this function); did you mean ‘float16_t’?
	Line  903: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:40:32: error: expected expression before ‘)’ token
	Line  906: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:42:19: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘*’ token
	Line  909: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:42:21: error: ‘w’ undeclared (first use in this function)
	Line  912: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:42:31: error: expected ‘)’ before ‘bfloat16_t’
	Line  935: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:51:13: error: unknown type name ‘bfloat16x8_t’
	Line  942: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:53:13: error: unknown type name ‘bfloat16x8_t’
	Line  945: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:54:13: error: unknown type name ‘bfloat16x8_t’
	Line  948: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:55:13: error: unknown type name ‘bfloat16x8_t’
	Line  951: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:56:13: error: unknown type name ‘bfloat16x8_t’
	Line  958: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:58:15: error: incompatible types when assigning to type ‘float32x4_t’ from type ‘int’
	Line  961: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:59:15: error: incompatible types when assigning to type ‘float32x4_t’ from type ‘int’
	Line  964: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:60:15: error: incompatible types when assigning to type ‘float32x4_t’ from type ‘int’
	Line  967: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:61:15: error: incompatible types when assigning to type ‘float32x4_t’ from type ‘int’
	Line  971: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:64:13: error: unknown type name ‘bfloat16x8_t’
	Line  974: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:64:59: error: expected ‘)’ before ‘bfloat16_t’
	Line  984: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:66:13: error: unknown type name ‘bfloat16x8_t’
	Line  987: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:67:13: error: unknown type name ‘bfloat16x8_t’
	Line  990: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:68:13: error: unknown type name ‘bfloat16x8_t’
	Line  993: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:69:13: error: unknown type name ‘bfloat16x8_t’
	Line 1001: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:71:40: error: incompatible type for argument 1 of ‘vceqq_u16’
	Line 1008: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:72:40: error: incompatible type for argument 1 of ‘vceqq_u16’
	Line 1015: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:73:40: error: incompatible type for argument 1 of ‘vceqq_u16’
	Line 1022: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:74:40: error: incompatible type for argument 1 of ‘vceqq_u16’
	Line 1029: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:76:13: error: unknown type name ‘bfloat16x8_t’
	Line 1036: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:76:67: error: incompatible type for argument 1 of ‘vbicq_u16’
	Line 1043: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:77:15: error: incompatible types when assigning to type ‘float32x4_t’ from type ‘int’
	Line 1046: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:78:13: error: unknown type name ‘bfloat16x8_t’
	Line 1049: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:78:67: error: incompatible type for argument 1 of ‘vbicq_u16’
	Line 1056: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:79:15: error: incompatible types when assigning to type ‘float32x4_t’ from type ‘int’
	Line 1059: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:80:13: error: unknown type name ‘bfloat16x8_t’
	Line 1062: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:80:67: error: incompatible type for argument 1 of ‘vbicq_u16’
[... cut ...]
	Line 1132: /arm_build/build/xnnpack/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c:119:88: error: incompatible type for argument 2 of ‘vext_u16’
	Line 1143: make[2]: *** [_deps/xnnpack-build/CMakeFiles/microkernels-all.dir/build.make:46951: _deps/xnnpack-build/CMakeFiles/microkernels-all.dir/src/bf16-gemm/gen/bf16-gemm-1x4c8-minmax-neonbf16-bfdot.c.o] Error 1
	Line 1145: make[1]: *** [CMakeFiles/Makefile2:6761: _deps/xnnpack-build/CMakeFiles/microkernels-all.dir/all] Error 2
	Line 1169: make: *** [Makefile:139: all] Error 2

These are the errors produced.

rascani, 2024-07-10:
This project is for TFLite for Microcontrollers, which is different than TFLite. I would suggest trying here: https://github.com/tensorflow/tensorflow/issues

closed 2024-07-10 by rascani