tensorflow/tflite-micro #2650: build: C++17 is not enabled on all platforms

rkuester, 2024-07-31:
Bazel builds have been C++17 since 52c9568; however, Makefile builds remain C++11. Consistent with our webpage documentation and to match TF Lite's language support, it's our desire to support and compile as C++17 on all platforms.

rkuester, 2024-07-31:
After trying to set -std=c++17 and c17 in PR #2648, the only problem appears to be the age of the Xtensa toolchain. The flatbuffers library includes `<optional>` when it's being compiled under the C++17 standard. The Xtensa toolchain is happy enough to compile with -std=c++17, but the standard library appears to be incomplete—it doesn't have `<optional>`. E.g., from the PR's CI run:

https://github.com/tensorflow/tflite-micro/actions/runs/10152485116/job/28074001989?pr=2648#step:3:367
````
xt-clang++ -std=c++17 [...]
[....]
.../micro/tools/make/downloads/flatbuffers/include/flatbuffers/stl_emulation.h:41:12: fatal error:
'optional' file not found
  #include <optional>
           ^~~~~~~~~~
````
Perhaps upgrading the Xtensa toolchain, which is quite old, will fix this.

For reference, the current toolchain is identified and invoked here: https://github.com/tensorflow/tflite-micro/blob/6b676567b79053d4d275ae834427b4f1259a710d/.github/workflows/xtensa_presubmit.yml#L35

rascani, 2024-07-31:
I tested this out for at least the hifi3 builds and got it to compile with `-stdlib=libc++`. I think the default stdlib for Xtensa only supports C++14. The downside is that I think that pre-compiled library includes rtti & exception support, so it likely has some bloat. In newer versions of Xtensa toolchain, we can use `libc++-re` to disable those.

rkuester, 2024-08-01:
Fixed in #2648.

closed 2024-08-01 by rkuester