tensorflow/tflite-micro #2676: Does TFLM support MFCC?

ctwillson, 2024-08-29:


shlmregev, 2024-09-09:
Not currently, but it shouldn't be too hard to add  support for it. With the signal/ ops you can implement a mel-spectrum filter bank followed by a log. To complete the MFCC computation, you'll need to add a DCT-II operation. Following the example code code in:
https://www.tensorflow.org/api_docs/python/tf/signal/mfccs_from_log_mel_spectrograms
You will need to add the operation at the very end:
mfccs = tf.signal.mfccs_from_log_mel_spectrograms(
  log_mel_spectrograms)[..., :13]

ctwillson, 2024-09-23:
> Not currently, but it shouldn't be too hard to add support for it. With the signal/ ops you can implement a mel-spectrum filter bank followed by a log. To complete the MFCC computation, you'll need [...]

Hi shlmregev：
This is a bit difficult for me. Will the official support MFCC? It is very effective for audio feature extraction

closed 2024-10-24 by github-actions[bot]