tensorflow/tflite-micro #2680: Fail to run person_detect.tflite via TFLite benchmark_model

bhbruce, 2024-09-06:
Link to issue in TensorFlow repo: https://github.com/tensorflow/tensorflow/issues/68308

### Issue type

Bug

### Have you reproduced the bug with TensorFlow Nightly?

Yes

### Source

source

### TensorFlow version

97a794b0199b6e507e3f577861856ff09357a3cd

### Current behavior?

There is a TFLite model called `person_detect.tflite` in tflite-micro https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/models/person_detect.tflite.
I cannot run this model via tensorflow's benchmark_model. 
It looks like all depthwise_convolutions' quantization_dim meta-data are wrong.

### Command to reproduce the issue

```shell
`tools/benchmark/benchmark_model  --num_runs=3 --enable_op_profiling=true --num_threads=2 --graph=person_detect.tflite`
```

### Relevant log output

```shell
### Error Log 

INFO: STARTING!
INFO: Log parameter values verbosely: [0]
INFO: Min num runs: [3]
INFO: Num threads: [2]
INFO: Graph: [person_detect.tflite]
INFO: Signature to run: []
INFO: Enable op profiling: [1]
INFO: #threads used for CPU inference: [2]
INFO: Loaded model person_detect.tflite
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 33 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 36 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 40 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 44 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 48 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 52 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 56 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 60 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 64 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 68 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 72 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 76 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 80 has invalid quantization parameters.
ERROR: quantized_dimension must be in range [0, 1). Was 3.
ERROR: Tensor 84 has invalid quantization parameters.
ERROR: Failed to initialize the interpreter
ERROR: Benchmarking failed.
```

bhbruce, 2024-09-06:
FYI. @sawantkumar

ddavis-2015, 2024-09-07:
@bhbruce This is the TFLM repo (aka LiteRT for Microcontrollers).  The issue you reference is for TFLite (aka LiteRT) repository which is currently the same as the tensorflow repo.

The quantization axis of 3 is correct for DEPTHWISE_CONV, and the person detection model in the TFLM repo. works as expected, passing all continuous integration tests.  The same model is used in the arduino-examples repo. and has been tested to work correctly on actual hardware.

This issue will now be closed.

closed 2024-09-07 by ddavis-2015