tensorflow/tflite-micro #2684: Adding support for Rfft2d

kirianguiller, 2024-09-16:
Hello everyone,

In the issue [Can this be done in TFLM](https://github.com/tensorflow/tflite-micro/issues/2478), it was mentionned that the kernels Rfft2d and ComplexAbs were not supported by TFLM. (kernels used in tflite for the audio classification model for instance).

I wanted to know if the reason of this is a technical incompatibility or if it's possible to implement by following the [step-by-step](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md) guide? 

Thanks!

rascani, 2024-09-23:
TFLM doesn't currently support complex types, which would be needed for both `Rfft2d` and `ComplexAbs`. We do have our own signal library (https://github.com/tensorflow/tflite-micro/tree/main/signal), but it does mean models using those operations need to be re-authored using the TFLM signal library.

HemanthSai7, 2024-10-09:
I too want to know if this op will be supported in future or it can be ported by following the guide.

closed 2024-11-08 by github-actions[bot]

Beyond9596, 2024-11-22:
> TFLM doesn't currently support complex types, which would be needed for both `Rfft2d` and `ComplexAbs`. We do have our own signal library [...]

I also want to know if TFLM will support complex types in the future? If not supported, what is the reason?