tensorflow/tflite-micro #2695: [Bug] Segmentation fault initializing model

Dimfred, 2024-09-24:
I am trying to load my model (the issue might be in the model itself). I think there is some bug here:
 https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_allocator.cc#L276
 
 So it seems that the segfault is called by trying to get element `0` from `zero_point`. I printed `src_quantiazation->zero_point()->size()` and it is `0`, hence probably the segfault.

Shouldn't there be a check making sure there exists a zero point before accessing it?

rascani, 2024-09-24:
Yes, there is a check on line 269: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_allocator.cc#L269

Not sure how you could wind up there with that conditional.

Dimfred, 2024-09-25:
Oh right, what the hell? I'll look deeper into it, but I am printing it and it is 0.

closed 2024-10-27 by github-actions[bot]