tensorflow/tflite-micro #2703: Tflite_micro wheel and bazel broken for aarch64

mansnils, 2024-09-27:
This PR https://github.com/tensorflow/tflite-micro/pull/2702 introduced a dependency to LiteRT.
It appears LiteRT does not support linux+aarch64 builds.

1) There's no aarch64 linux wheel for ai-edge-litert
2) It appears not even possible to build natively for aarch64 linux from source either (using the ci/build_pip_package_with_bazel.sh script): https://github.com/google-ai-edge/LiteRT

This is a blocker if you are using Linux+Aarch64.
Until this is resolved it would be good to revert the PR.

closed 2024-09-27 by rascani

rkuester, 2024-09-27:
For what it's worth, this also breaks the build on machines running Python 3.12. LiteRT's Python packages only support up to Python 3.11. Python 3.12 was released one year ago---October 2023. Tensorflow proper has supported 3.12 since at least November 2023.

rkuester, 2024-09-27:
> For what it's worth, this also breaks the build on machines running Python 3.12. LiteRT's Python packages only support up to Python 3.11. Python 3.12 was released one year ago---October 2023. Tensorflow proper has supported 3.12 since at least November 2023.

A 3.12 package is available in [ai-edge-litert-nightly](https://pypi.org/project/ai-edge-litert-nightly/), so I expect it's coming to [ai-edge-litert](https://pypi.org/project/ai-edge-litert/) soon.