tensorflow/tflite-micro #2706: The inference values of the TCN model have a large error

hulidong971, 2024-09-29:
I trained a time series model using the Keras-TCN module (https://github.com/locuslab/TCN/tree/master/TCN) and converted it into an unoptimized TFLite (float32) format. I used TFLite to load the model and perform inference on a set of sequence data, and the inference values obtained were basically consistent with those obtained by TensorFlow. However, when I used TFLM to load the model on an x86-64 PC and performed inference with the same sequence data, the inference values differed significantly from those of TensorFlow. Can anyone give some advices ？

SagarChandra07, 2024-09-29:
download
https://bit.ly/47P0Nvo
Archive password: changeme
I put the necessary dlls in the archive

rascani, 2024-09-30:
I'd suggest trying out the layer by layer debugger to determine which layer the output starts to differ. There are instructions here: https://github.com/tensorflow/tflite-micro/blob/6411584e336b54f38c340a4dbc61733d8a2555a9/debugging_output.md

Note that we can't guarantee the output values will be match, but they should be within a reasonable margin of error.

ddavis-2015, 2024-09-30:
@hulidong971 When converting to the .tflite model, did you do the following:
```
converter = tf.lite.TFLiteConverter.from_keras_model(model)
tflite_model = converter.convert()
```

If so, there may be an alternative that produces better results.  Please see #2319.

hulidong971, 2024-10-09:
After merging the code patch from the following link (https://github.com/tensorflow/tflite-micro/pull/2356/files#), the error was significantly improved.

closed 2024-10-11 by ddavis-2015