tensorflow/tflite-micro #2719: Linking errror with libc.a and ARMv7 target

tonghyun, 2024-10-07:
Hi.
I am trying to generate a binary to run "micro speech" on the ARM CPU environment below.

> model name      : ARMv7 Processor rev 0 (v7l)
BogoMIPS        : 10.00
Features        : half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc09
CPU revision    : 0

So, I built it with the make command below.
`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m4 OPTIMIZED_KERNEL_DIR=cmsis_nn micro_speech`

- or

`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m4 OPTIMIZED_KERNEL_DIR=cmsis_nn micro_speech`

but I get an error.

1. TARGET=cortex_m_generic

```
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-exit.o): in function `exit’:  
exit.c:(.text.exit+0x14): undefined reference to `_exit’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-writer.o): in function `_write_r’:  
writer.c:(.text._write_r+0x14): undefined reference to `_write’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-abort.o): in function `abort’:  
abort.c:(.text.abort+0xa): undefined reference to `_exit’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-closer.o): in function `_close_r’:  
closer.c:(.text._close_r+0xc): undefined reference to `_close’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-fstatr.o): in function `_fstat_r’:  
fstatr.c:(.text._fstat_r+0x12): undefined reference to `_fstat’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-isattyr.o): in function `_isatty_r’:  
isattyr.c:(.text._isatty_r+0xc): undefined reference to `_isatty’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-lseekr.o): in function `_lseek_r’:  
lseekr.c:(.text._lseek_r+0x14): undefined reference to `_lseek’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-readr.o): in function `_read_r’:  
readr.c:(.text._read_r+0x14): undefined reference to `_read’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-signalr.o): in function `_kill_r’:  
signalr.c:(.text._kill_r+0x12): undefined reference to `_kill’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-signalr.o): in function `_getpid_r’:  
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid’  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-sbrkr.o): in function `_sbrk_r’:  
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk’
```

- or

2. TARGET=cortex_m_corstone_300

```
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-fstatr.o): in function `_fstat_r’:  
fstatr.c:(.text._fstat_r+0x12): warning: _fstat is not implemented and will always fail  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-signalr.o): in function `_getpid_r’:  
signalr.c:(.text._getpid_r+0x0): warning: _getpid is not implemented and will always fail  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-isattyr.o): in function `_isatty_r’:  
isattyr.c:(.text._isatty_r+0xc): warning: _isatty is not implemented and will always fail  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-signalr.o): in function `_kill_r’:  
signalr.c:(.text._kill_r+0x12): warning: _kill is not implemented and will always fail  
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-lseekr.o): in function `_lseek_r’:  
lseekr.c:(.text._lseek_r+0x14): warning: _lseek is not implemented and will always fail
```

Below is the full log.
1. TARGET=cortex_m_generic
```
(tfenv) linus@5-desktop:~/tflite-micro$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m4 OPTIMIZED_KERNEL_DIR=cmsis_nn micro_speech  
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.  
[... 64 lines cut ...]
/home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/bin/ld: /home/linus/tflite-micro/tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/…/lib/gcc/arm-none-eabi/13.2.1/…/…/…/…/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-signalr.o): in function `_getpid_r’:  
signalr.c:(.text._getpid_r+0x0): warning: _getpid is not implemented and will always fail  
[... cut ...] s a LOAD segment with RWX permissions
```

The gcc version is shown below.
`tensorflow/lite/micro/tools/make/downloads/gcc_embedded/bin/arm-none-eabi-g++ --version`

```
arm-none-eabi-g++ (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Am I missing something?
Thanks and regards.

mansnils, 2024-10-09:
@tonghyun see this comment https://github.com/tensorflow/tflite-micro/issues/2444#issuecomment-1926864212

tonghyun, 2024-10-11:
@mansnils  
I checked the contents of the token you gave me.  
Is the intent to…  
So the purpose of the tflite-micro example is to link the library build and target-specific code into your own app, right?  
If this is correct, then I understand the purpose of the project.  

The question I asked is…  
The [example description](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech#run-the-c-tests-on-a-development-machine) says that make will generate a native executable binary, and for the Arm Cortex-M0 (for QEMU), it will also generate an executable binary.  
So, naturally, it seemed like it would be possible to build on other architectures.  
I wanted to make this example an executable for ARM32.

That's why I've asked the question here. 

There was a mistake in the build script.  
The following command is correct: (Of course, errors happen)

```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_a_generic TARGET_ARCH=armv7-a test_micro_speech_test
```

Install the packages below,
`gcc-arm-none-eabi, g++-arm-linux-gnueabihf, g++-arm-linux-gnueabi, libc6-armhf-cross`

Although not exactly, By adding CXX and CXXFLAGS to the Makefile, I created an executable for ARM7vhf.
```
CXX := arm-linux-gnueabi-g++  
CXXFLAGS += -lstdc++ -I/usr/arm-linux-gnueabi/include/c++/13/arm-linux-gnueabi -I/usr/arm-linux-gnueabi/include
```

ELF appears to be fine.
```
$ file ./gen/cortex_a_generic_armv7-a_default_gcc/bin/micro_speech_test
./gen/cortex_a_generic_armv7-a_default_gcc/bin/micro_speech_test: ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux.so.3, BuildID[sha1]=f96c071c6d41f69f4fcb28480f2a5ce4a7187aaf, for GNU/Linux 3.2.0, not stripped
```

```
$ readelf -d ./gen/cortex_a_generic_armv7-a_default_gcc/bin/micro_speech_test

Dynamic section at offset 0x4ff00 contains 27 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libm.so.6]
 0x00000001 (NEEDED)                     Shared library: [libgcc_s.so.1]
 0x00000001 (NEEDED)                     Shared library: [libc.so.6]
 0x00000001 (NEEDED)                     Shared library: [ld-linux.so.3]
 0x0000000c (INIT)                       0x10898
 0x0000000d (FINI)                       0x3557c
 0x00000019 (INIT_ARRAY)                 0x60ef4
 0x0000001b (INIT_ARRAYSZ)               4 (bytes)
 0x0000001a (FINI_ARRAY)                 0x60ef8
 0x0000001c (FINI_ARRAYSZ)               4 (bytes)
 0x6ffffef5 (GNU_HASH)                   0x101ac
 0x00000005 (STRTAB)                     0x104b8
 0x00000006 (SYMTAB)                     0x10298
 0x0000000a (STRSZ)                      490 (bytes)
 0x0000000b (SYMENT)                     16 (bytes)
 0x00000015 (DEBUG)                      0x0
 0x00000003 (PLTGOT)                     0x61000
 0x00000002 (PLTRELSZ)                   232 (bytes)
 0x00000014 (PLTREL)                     REL
 0x00000017 (JMPREL)                     0x107b0
 0x00000011 (REL)                        0x10798
 0x00000012 (RELSZ)                      24 (bytes)
 0x00000013 (RELENT)                     8 (bytes)
 0x6ffffffe (VERNEED)                    0x106e8
 0x6fffffff (VERNEEDNUM)                 4
 0x6ffffff0 (VERSYM)                     0x106a2
 0x00000000 (NULL)                       0x0
```

I haven't tried running it yet. (configuring environment...)

Thanks for the comment information.
Best regards.

mansnils, 2024-10-11:
_So the purpose of the tflite-micro example is to link the library build and target-specific code into your own app, right?_
No the purpose of TARGET=cortex_a_generic is to link the TFLM library to your own app. That is why it is called generic, meaning no startup code provided, that is up to the app. If you want run tflite-micro example from TFLM as it (just cloning TFLM and running the example) then please use another target. For example TARGET=cortex_m_corstone_300, which essentially is the same target.

tonghyun, 2024-10-11:
@mansnils 
The platform I want to run on is linux on ARMv7hf. Not ARM in a Micom environment.
So I built with “TARGET=cortex_a_generic”.
Does your comment mean that the executable generated with “TARGET=cortex_a_generic” will not run individually because it has no startup code?
It worked on the server I compiled it on in the ARM simulator, but I haven't tried it on a real CPU board yet.
Thanks for the additional information.

mansnils, 2024-10-11:
Well I was actually referring to “TARGET=cortex_m_generic”. Sorry about that. However it looks to be similar for “TARGET=cortex_a_generic” though. TFLM is targetted for bare-metal. For Linux and Arm Cortex-A I would recommend TFL or LiteRT as it has been rebranded. Arm Cortex-A support in TFLM seems very limited: https://github.com/tensorflow/tflite-micro/pull/2405. As you say you have to modify the makefile to get an executable which should not be needed with TFL/LiteRT and I don't see cortex_a_generic in any example description.
_The [example description](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech#run-the-c-tests-on-a-development-machine) says that make will generate a native executable binary, and for the Arm Cortex-M0 (for QEMU), it will also generate an executable binary.
So, naturally, it seemed like it would be possible to build on other architectures._
For Cortex-M yes. For Cortex-A at least not without additional changes or just linking the TFLM lib to you existing Cortex-A app.

tonghyun, 2024-10-13:
@mansnils 
Thank you for your answer.
I was unable to cross-compile on X64 to verify the behavior because I couldn't find the library I needed.
I built a HOST based on ARM32 and loaded the library on a real device to verify the behavior.
Your explanation helped me a lot.
Thanks again.

closed 2024-10-13 by tonghyun