tensorflow/tflite-micro #2720: Add INT32 Support to Sub

HemanthSai7, 2024-10-09:
When trying to deploy the model on ESP32S3, I get a warning called `SUB failed to invoke INT32 not supported`. I have fixed it and I want to contribute by opening a PR.

ddavis-2015, 2024-10-09:
@HemanthSai7 

TFLM (aka LRTM) welcomes PR contributions!  Please submit your PR at your leisure with these rules in mind:
- All code must comply with Google C++ style rules
- All added files (if any) must build through both Makefile and Bazel
- Unit tests that allow complete code coverage must be submitted

So you would need unit tests in sub_test.cc for INT32 that cover both broadcast and non-broadcast (just like the INT16).  Also be aware that if this is your first time submitting code, you will need to sign the CLA.  You will be presented with information on how to do this when you submit the PR.

HemanthSai7, 2024-10-12:
Okay, I will keep these points in mind.

HemanthSai7, 2024-10-22:
Hello, for opening the PR I have to follow the steps mentioned in Contributing.md

```bash
git remote add HemanthSai7 git@github.com:HemanthSai7/tflite-micro.git
git fetch HemanthSai7

git checkout -b HemanthSai7-tflm-int32-sub-support HemanthSai7/<PR branch name>

# make changes and commit to local branch

# push changes to remove branch

git push HemanthSai7 <PR branch name>

# remove the temp remote to clean up your git environment.

git remote rm HemanthSai7
```

What is the PR branch name?

HemanthSai7, 2024-11-20:
This issue is not stale. I will create a PR.

HemanthSai7, 2025-01-09:
How to reopen this issue?

PR #3037 (2025-06-04): Add INT32 support to SUB
- Add INT32 support in sub
- Add Tflite tests in sub_test.cc

bug=fixes #2720
commit 686df2db (2025-06-04): Add INT32 support to SUB (#3037)

- Add INT32 support in sub
- Add Tflite tests in sub_test.cc

bug=fixes #2720
closed 2025-06-04 by mergify[bot]