tensorflow/tflite-micro #2721: Unhandled overflow in int8 SVDF evaluation

VictorDominguite, 2024-10-09:
When calculating the values to be inserted in the SVDF operator’s activation state in the [svdf_common.cc](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/svdf_common.cc) file, there seems to be an error while handling overflow.
In [line 111](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/svdf_common.cc#L111) the value for the dot_prod variable is limited to the interval of valid int8 values, [-128, 127]. But then, in [line 115](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/svdf_common.cc#L115), the activation state’s zero point is added to dot_prod, which may lead to an unhandled overflow.

renatocan, 2024-11-11:
Commenting to prevent closing the issue.

My colleagues @VictorDominguite and Luiz Baptista have proposed a potential solution for this issue. We'll create a PR to further discuss and implement it. Thanks!

renatocan, 2024-12-09:
Commenting to prevent closing the issue.