tensorflow/tflite-micro #2730: [REQUEST] Please add port TensorFlow Lite Micro to the new RP2350 (and old RP2040) microcontroller platforms (e.g. Raspberry Pi Pico 2)?

Hedda, 2024-10-25:
Can you please add an official port of the TensorFlow Lite Micro library to new RP2350 (and old RP2040) microcontroller platforms?

There looks to be a few unofficial ports of the [TensorFlow Lite Micro library](https://www.tensorflow.org/lite/microcontrollers) for the RP2040 but obviously RP2350 should perform better.

There was LiteRT downstream port for older RP2040 but do not look to be maintained and has not been updated to new RP2350, see:

* https://www.raspberrypi.com/news/a-big-bang-update-for-tensorflow-lite-for-microcontrollers/

and

* https://github.com/raspberrypi/pico-tflmicro

RP2350 is the successor and next-generation of RPi RP2040 so guess ESPHome would need to update the [RP2040 Platform](https://esphome.io/components/rp2040.html) for it or add another platform?

* https://en.wikipedia.org/wiki/RP2350
  * https://www.raspberrypi.com/products/rp2350/

and/or

* https://en.wikipedia.org/wiki/RP2040
  * https://www.raspberrypi.com/products/rp2040/

RP2350 is newly released so there is not yet widespread support but  RP2040 based development boards are already very popular in schools and STEM education so think they are interesting as alternatives to ESP32 because documentation is often easier as often targeting beginners which in turn means can lead to broader use and use cases.

FYI, there is not yet an official Raspberry Pi Pico 2 board with Wi-Fi but there are third-party boards with WiFi and LAN, see:

* https://www.raspberrypi.com/news/our-rp2350-partners-made-all-this-excellent-stuff-for-you/
  * https://www.raspberrypi.com/for-industry/powered-by/product-catalogue/?category=RP2350
    * https://ilabs.se/?s=RP2350&post_type=product
    * https://docs.wiznet.io/Product/iEthernet/W6100/w6100-evb-pico2

Hedda, 2024-10-25:
If added it by the way would be very cool if could get Raspberry Pi to then make a demo video showing microWakeWord running natively on RP2350 or RP2040 microcontrollers and maybe post a blog post about it a show case for [tflite-micro](https://github.com/tensorflow/tflite-micro) and Raspberry Pi Pico 2!

* https://github.com/kahrendt/microWakeWord

microWakeWord (also known as "mWW") is a fully open-source wake word detection library made for MCU-based voice assistants, and it depends on the [TensorFlow Lite (LiteRT) library](https://www.tensorflow.org/lite/microcontrollers) to run fast nativley on low-power microcontrollers:

* https://www.kevinahrendt.com/micro-wake-word

* https://www.youtube.com/watch?v=oSKBWtBJyDE

The [microWakeWord](https://github.com/kahrendt/microWakeWord) was created by [Kevin Ahrendt](https://www.kevinahrendt.com/) and it enables for example the ESPHome firmware (via microphones) to on-device wake words detections on microcontroller devices.

* https://www.home-assistant.io/voice_control/about_wake_word/#about-on-device-wake-word-processing-microwakeword

* https://esphome.io/components/micro_wake_word.html

* https://www.home-assistant.io/voice_control/about_wake_word/

* https://www.home-assistant.io/blog/2024/06/26/voice-chapter-7/#big-things-for-microwakeword

* https://www.home-assistant.io/blog/2024/06/26/voice-chapter-7/#microwakeword-and-kevin-join-the-team

If you do please post a new blog post to announce the updated and added RP2350 support!

* https://www.raspberrypi.com/news/a-big-bang-update-for-tensorflow-lite-for-microcontrollers/

PS: The microWakeWord  project as been adopted/transfered to the Open Home Foundation so it will always remain as open-source:

* https://www.cnx-software.com/2024/08/01/the-open-home-foundation-adds-hacs-microwakeword-and-music-assistant-open-source-projects/

* https://www.hackster.io/news/the-open-home-foundation-grows-as-it-names-hacs-music-assistant-and-microwakeword-as-partners-2f60ac645e8f

* https://newsletter.openhomefoundation.org/the-open-home-is-not-for-sale/

Hedda, 2024-11-19:
Still wanted.

Hedda, 2024-12-16:
This is still wanted and would think this be of interest since it is a very popular development platform for new developers and beginners starting with microcontrollers as much used in schools or other classes and other education situations, (i.e. RP2040 and RP2350 will in the future likely be one of the more common microcontroller development platforms which many of the next-generation developers got started on).

petewarden, 2024-12-31:
I believe this is a duplicate of https://github.com/raspberrypi/pico-tflmicro/issues/18. Since I've now implemented RP2350 support on that downstream project, I believe this bug can be closed (unless I'm missing something). 

Platform support for TFLM is generally expected to be the responsibility of platform owners, hence the project existing in the RPi organization, so I don't think the Google team will take this one up themselves.

closed 2025-01-31 by github-actions[bot]