tensorflow/tflite-micro #2748: Linking the tflite micro library in PX4

SindreMHegre, 2024-11-04:
I have built a .a library and copied all the header files over. But when I try to build PX4 I get errors in what I believe to be the linking process. I'm not very experienced with building. Should I maybe change the CMakelists.txt file in this directory: https://github.com/SindreMHegre/PX4-Autopilot/tree/pr_nn_embedded_control_1.15/src/lib/tflite

Or would it be easier to add the entire codebase of tflite-micro as a submodule?

Here is the end of the console printout when trying to build now:
`link px4 px4-px4_simple_app && cd /home/sindre/***/PX4-Autopilot/build/px4_sitl_default/bin && /usr/bin/cmake -E create_symlink px4 px4-work_item_example
src/modules/mc_nn_control/libmc_nn_control.a(mc_nn_control.cpp.o):mc_nn_control.cpp:function mc_nn_control_main.cold: error: undefined reference to 'tflite::MicroInterpreter::~MicroInterpreter()'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:46: error: undefined reference to 'tflite::Register_FULLY_CONNECTED()'
/home/sindre/**/dev/PX4-Autopilot/src/lib/tflite/tensorflow/lite/micro/micro_mutable_op_resolver.h:681: error: undefined reference to 'MicroPrintf(char const*, ...)'
/home/sindre/**/PX4-Autopilot/src/lib/tflite/tensorflow/lite/micro/micro_mutable_op_resolver.h:682: error: undefined reference to 'MicroPrintf(char const*, ...)'
/home/sindre/**/PX4-Autopilot/src/lib/tflite/tensorflow/lite/micro/micro_mutable_op_resolver.h:693: error: undefined reference to 'tflite::ParseFullyConnected(tflite::Operator const*, tflite::ErrorReporter*, tflite::BuiltinDataAllocator*, void**)'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:71: error: undefined reference to 'tflite::MicroInterpreter::MicroInterpreter(tflite::Model const*, tflite::MicroOpResolver const&, unsigned char*, unsigned long, tflite::MicroResourceVariables*, tflite::MicroProfilerInterface*, bool)'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:74: error: undefined reference to 'tflite::MicroInterpreter::AllocateTensors()'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:81: error: undefined reference to 'tflite::MicroInterpreter::input(unsigned long)'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:93: error: undefined reference to 'tflite::MicroInterpreter::Invoke()'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:100: error: undefined reference to 'tflite::MicroInterpreter::output(unsigned long)'
/home/sindre/**/PX4-Autopilot/src/modules/mc_nn_control/mc_nn_control.cpp:112: error: undefined reference to 'tflite::MicroInterpreter::~MicroInterpreter()'
/home/sindre/**/PX4-Autopilot/src/lib/tflite/tensorflow/lite/micro/micro_mutable_op_resolver.h:675: error: undefined reference to 'MicroPrintf(char const*, ...)'
/home/sindre/**/PX4-Autopilot/src/lib/tflite/tensorflow/lite/micro/micro_mutable_op_resolver.h:676: error: undefined reference to 'MicroPrintf(char const*, ...)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1`

SindreMHegre, 2024-11-04:
I think I was able to link it properly. But when I run my module I get this error:

Didn't find op for builtin opcode 'ADD'
Failed to get registration from op code ADD
 
ERROR [mc_nn_control] AllocateTensors() failed

And here is the code until that point:
`namespace {
using NNControlOpResolver = tflite::MicroMutableOpResolver<2>;

TfLiteStatus RegisterOps(NNControlOpResolver& op_resolver) {
  TF_LITE_ENSURE_STATUS(op_resolver.AddFullyConnected());
  return kTfLiteOk;
}
}  // namespace

extern "C" __EXPORT int mc_nn_control_main(int argc, char *argv[])
{
	PX4_INFO("Hello from mc_nn_control_main");

	// Load the model
	const tflite::Model* model = ::tflite::GetModel(simple_net_tflite);  // Replace with your model data variable
	if (model->version() != TFLITE_SCHEMA_VERSION) {
		PX4_ERR("Model provided is schema version %d not equal to supported version %d.",
			model->version(), TFLITE_SCHEMA_VERSION);
		return -1;
	}

	// Set up the interpreter
	static NNControlOpResolver resolver;
	if (RegisterOps(resolver) != kTfLiteOk) {
		PX4_ERR("Failed to register ops");
		return -1;
	}
	constexpr int kTensorArenaSize = 10 * 1024;
	static uint8_t tensor_arena[kTensorArenaSize];
	tflite::MicroInterpreter interpreter(model, resolver, tensor_arena, kTensorArenaSize);

	// Allocate memory for the model's tensors
	TfLiteStatus allocate_status = interpreter.AllocateTensors();
	if (allocate_status != kTfLiteOk) {
		PX4_ERR("AllocateTensors() failed");
		return -1;`

ddavis-2015, 2024-11-04:
@SindreMHegre 

You will need to register all operators your model uses:
```
TF_LITE_ENSURE_STATUS(op_resolver.AddFullyConnected());
TF_LITE_ENSURE_STATUS(op_resolver.AddAdd());
```

This issue will now be closed.  Please feel free to reopen the issue should you have additional comments.

closed 2024-11-04 by ddavis-2015