tensorflow/tflite-micro #3030: section `.data' can't be allocated in segment 0 LOAD: .text .ARM.exidx .copy.table .zero.table .data

asyatrhl, 2025-01-03:


ArmRyan, 2025-01-22:
Hi @asyatrhl This is not really an "error", more of a weird implementation warning I guess. It should still work even with this message.

If you don't find thats the case then can you provide any further info on where it fails?
The output of that make command should still be the hello world binary

mansnils, 2025-02-03:
Hi @asyatrhl, I think hello_world will just build the binary and test_hello_world will actually run the test. Is it working for you?

asyatrhl, 2025-02-10:
Hi @mansnils, when I try to build the hello_world example for arm_cortex_m4 (did not change anything on the cloned repository) the above error still occurs.  When I try to build hello_world example for riscv there is no such warning.
I want to to build hello_world example and flash it to arm_cortex_m4 on MAX78000 chip. 
When I try to flash there is also another error occurs which I think depends on this. Warning: Loadable section ".data" outside of ELF segments The target architecture is set to "armv7e-m".  (I am using my own linker file for this step.)

mansnils, 2025-02-10:
@asyatrhl as Ryan mentioned the error msg is just a warning. I was able to run your command:
```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m4 OPTIMIZED_KERNELS=cmsis_nn hello_world
echo $?
0
```
Now if you are using your own linker script I can't reproduce or guarantee it is working.
Note also if you want to flash your own HW don't use TARGET=cortex_m_corstone_300 - instead use TARGET=cortex_m_generic. Also not for TARGET=cortex_m_generic you can only build the tflite micro lib and then link it to your application:
`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_generic TARGET_ARCH=cortex-m4 OPTIMIZED_KERNELS=cmsis_nn microlite`

asyatrhl, 2025-02-10:
@mansnils When I do not use my own linker there is an error "section `.data' can't be allocated in segment 0 LOAD: .text .ARM.exidx .copy.table .zero.table .data". (as I explained before, i do not change anything, and just run  "make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m4 OPTIMIZED_KERNELS=cmsis_nn hello_world").

mansnils, 2025-02-16:
Hi @asyatrhl 
What happens if you run:
`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m4 OPTIMIZED_KERNELS=cmsis_nn hello_world`
And then immediately after run:
`echo $?`
Will it return 0 or not? If not it may be the build command is not working with your own linker.