tensorflow/tflite-micro #3055: Fixed-point Softmax() calls exp_on_negative_values() twice

cousteaulecommandant, 2025-02-27:
Fixed-point `Softmax()` calls `exp_on_negative_values()` twice per element: once to calculate the value of the exp of each element
https://github.com/tensorflow/tflite-micro/blob/ef64591270691022a329cf04ba9e73ecfb15ddb8/tensorflow/lite/kernels/internal/reference/softmax.h#L131
and another one a few lines earlier to calculate the sum of all exps
https://github.com/tensorflow/tflite-micro/blob/ef64591270691022a329cf04ba9e73ecfb15ddb8/tensorflow/lite/kernels/internal/reference/softmax.h#L109-L110
The `exp_on_negative_values()` function is rather slow, taking most of the time of `Softmax()`, so calling it twice per element makes the whole `Softmax()` function be about twice as slow as necessary.

Could this be optimized so that it is only called once per element?  Store the exp results in a temporary array (for example, reusing `output_data`), computing the sum as they're stored, and then applying the scaling to the precomputed exp results rather than computing them again.

ddavis-2015, 2025-03-04:
@cousteaulecommandant Yes, this does seem inefficient.  However, this file (along with most reference operators) are *imported* into TFLM (LRTM) from TfLite (LiteRT).  You would need to post this issue into the [tensorflow](https://github.com/tensorflow/tensorflow/issues) repository, in order for a change to be considered.

I would note that it appears the INT16 SOFTMAX operator does appear to have an optimization similar to what you propose.  Perhaps a CAST operation to/from INT16 would work for you?  INT16 SOFTMAX and CAST are supported in TFLM, though you would add the cost of a memory copy operation for each CAST.

This issue will now be closed.  Please feel free to reopen the issue should you have additional further comments.

closed 2025-03-04 by ddavis-2015

cousteaulecommandant, 2025-03-07:
Created tensorflow/tensorflow#88861