tensorflow/tflite-micro #3068: How to Split a tflite Model into N Blocks on ARM Cortex M55 to Reduce Memory Usage?

OmnishamblesOUO, 2025-03-26:
Background: 
I want to infer the model.tflite on ARM Cortex M55, but due to limited hardware memory, I plan to split the model into 4 blocks and execute them sequentially. The model has a single input/output structure, making it suitable for split-and-infer.

Questions:
1. Are there any APIs available to split model.tflite into 4 blocks?
2. I attempted to split the model via ONNX, then quantized and converted it to 4 block.tflite files using onnx2tf. However, during inference, the memory usage (interpreter.arena_used_bytes()) per block did not decrease. What could be the reason?
3. In TFLite-Micro inference, does tensor_arena size need to account for all weights + all intermediate variables (e.g., feature maps), or just the weights + largest intermediate variable?

Thanks!

closed 2025-03-26 by ddavis-2015

ddavis-2015, 2025-03-31:
@OmnishamblesOUO An incorrect statement was made in reply to your 2nd question.  I have made corrections to my previous answer.  However, those corrections do not, I think, bring you closer to your intended goal.

OmnishamblesOUO, 2025-04-01:
> [@OmnishamblesOUO](https://github.com/OmnishamblesOUO) An incorrect statement was made in reply to your 2nd question. I have made corrections to my previous answer. However, those corrections do not, I think, bring you closer to your intended goal.

Thank you for your reply and assistance!
I have resolved issues 2 and 3 and now understand the general reasons:

The memory did not decrease because the feature map of block1 was very large, which required a larger tensor_arena allocation, leading me to mistakenly believe the memory usage had not reduced.

Similarly, I now understand your explanation regarding issue 3.

Once again, thank you for your help!

However, regarding the first point, I still have some questions.

My goal is to split an int8 quantized tflite model into 4 parts and perform sequential inference on a Cortex M55.
The issue I’m currently facing is that I still don’t know how to split the tflite model into 4 parts. I’d like to seek your guidance on whether there is an official method for splitting tflite models or any workaround solutions to achieve this indirectly.

I split the model into 4 blocks using ONNX and then performed post-quantization and model conversion via onnx2tflite. However, the resulting tflite model produces incorrect results during sequential inference—no matter the input image, the output is always a fixed value (e.g., [[117]]). Do you have any related experience or suggestions for addressing this issue?

Thank you!

ddavis-2015, 2025-04-04:
@OmnishamblesOUO There is no TFLM tool for splitting a TfLite model file.  I also have not heard of such a tool.

As for your model(s) not generating the correct results, you could try the [layer-by-layer debugging tool](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/layer_by_layer_debugger.py).  If both the TfLite and TFLM results are wrong, then it is likely the model conversion had a problem.  If only the TFLM results are wrong, it could be an operator in the model is being used in a way not supported by TFLM.