tensorflow/tflite-micro #3070: Reshape Node failed with num_input_elements != num_output_elements (16000 != 32)

xavierfortier1, 2025-03-31:
I am trying to implement a model into a nRF5340 with their SDK. Everything was compiling and running until we tried to change the input shape. The hello world sample worked great and our model too. Our input was previously an array of shape 1x4, but we changed it to have 500x4. I have notice that doing 500 x 32 = 16 000. So, at first it seems that the implementation of the lib is not handling well arrays with arbitrary size. In our python model implementation and training, we are telling to the model that the `inputs_shape` is 500x4. I am therefore seeking help to resolve that problem. I have notice other issues with the problem, but none seems to be actually completed. 

This is the output we get when we are using our new model:

```
WEST_TOPDIR/optional/modules/lib/tflite-micro/tensorflow/lite/micro/kernels/reshape_common.cc:78 num_input_elements != num_output_elements (16000 != 32)
WEST_TOPDIR/optional/modules/lib/tflite-micro/tensorflow/lite/micro/kernels/reshape_common.cc:78 num_input_elements != num_output_elements (16000 != 32)
WEST_TOPDIR/optional/modules/lib/tflite-micro/tensorflow/lite/micro/kernels/reshape_common.cc:90 ReshapeOutput(context, node) != kTfLiteOk (1 != 0)
Node RESHAPE (number 7f) failed to prepare with status

AllocateTensors() failed (err 1)
```

ddavis-2015, 2025-03-31:
@xavierfortier1 TFLM (aka LRTM) does not support resizing of tensors.  This is because TFLM does not support dynamic memory allocation.  TFLM does support reshaping of tensors, provided the old and new tensor shapes result in a tensor of unchanged overall size.

I would suggest visualizing your model in order to make the proper adjustments to your model, prior to conversion to the TfLite model format.

- Using the Tensorflow Model Explorer ([Colab](https://colab.research.google.com) example):
```
!pip install --no-deps ai-edge-model-explorer-adapter ai-edge-model-explorer

import model_explorer

model_path = "model.tflite"
model_explorer.visualize(model_path)
```
- Using the experimental Tflite model analyzer ([Colab](https://colab.research.google.com) example):
```
import tensorflow as tf

tf.lite.experimental.Analyzer.analyze(model_path="model.tflite")
```

- Using an online web application such as [Netron](https://netron.app).

This issue will now be closed.  Please feel free to reopen this issue should you have additional comments.

closed 2025-03-31 by ddavis-2015

xavierfortier1, 2025-03-31:
@ddavis-2015 I was wondering if you had any idea how we could adjust our LSTM model made with Keras, so that it could work with `tflite-micro`.

ddavis-2015, 2025-04-01:
@xavierfortier1 Did you build/train your model using Keras **and** use `tf.lite.TFLiteConverter.from_keras_model` **and** using one of the inspection methods above, find that the RESHAPE operation was attempting to change the tensor size?

xavierfortier1, 2025-04-01:
@ddavis-2015 We've built our model with Keras, used `tf.lite.TFLiteConverter.from_keras_model` and used Netron, and there's a RESHAPE that modifies the size of the tensor.

ddavis-2015, 2025-04-01:
@xavierfortier1 Can you post an image showing just the Netron node properties for the RESHAPE operator?  Please make sure all expand all the inputs and outputs in the node properties window.

xavierfortier1, 2025-04-01:
@ddavis-2015 Here is the reshape node properties

[image]

ddavis-2015, 2025-04-01:
@xavierfortier1 The `reshaped` tensor has an illegal shape in TFLM (two -1 values), and should in fact have generated a different error message than the one you originally reported.

You can try converting the model using something similar to the snippet below, and see if the output shape is correct.  If not, it will be up to you to redo your model such that the RESHAPE (whether part of your original model or generated by Keras), has a correct output shape).  TFLM simply does not support tensor resizing.

```
# Create example dilated convolutional model
inputs = tf.keras.layers.Input(shape=INPUT_SHAPE)
y = inputs
y = tf.keras.layers.Convolution1D(1, 3, dilation_rate=DILATION_RATE, padding='same')(y)

# Create Keras model and concrete function
model = tf.keras.Model(inputs=inputs, outputs=y)
model_func = tf.function(func=model)
cf = model_func.get_concrete_function(
          tf.TensorSpec(shape=(1,)+INPUT_SHAPE, dtype=tf.float32))

# Convert to TF Lite model
converter = tf.lite.TFLiteConverter.from_concrete_functions([cf], model)
tflite_model = converter.convert()
```

xavierfortier1, 2025-04-02:
@ddavis-2015 Thanks for your help! We'll try that and change our model if it doesn't work